nordicsemi/IOS-DFU-Library #370: 4.7.1 carthage issue has no shared framework schemes

AlexDM0, 2020-04-21:
When installing the 4.7.1 version from Carthage (V0.34.0) 
```
github "NordicSemiconductor/IOS-Pods-DFU-Library" == 4.7.1
```
I get the following error:
*** Skipped building IOS-Pods-DFU-Library due to the error:
Dependency "IOS-Pods-DFU-Library" has no shared framework schemes

I can't really find any reason why it shouldn't work by looking at the releases and their contents... Maybe you guys have a suggestion?

diamondhnds, 2020-04-23:
We are seeing the same issue

philips77, 2020-04-24:
I'll look into this.

philips77, 2020-04-24:
https://github.com/Carthage/Carthage/pull/2965

philips77, 2020-04-24:
Did you try previous version of dfu library, or older Carthage version?

AlexDM0, 2020-04-24:
I have used 4.2.2 for a long time with Carthage. That worked fine. I figured I might upgrade now that it’s more than a year old.

philips77, 2020-05-13:
Hi, did you manage to solve the issue?
I'm about to release a new version and want to know how to fix Carthage configuration.

philips77, 2020-05-13:
Ok, I manager to reproduce it as well. Will work on it now.

AlexDM0, 2020-05-13:
I went back to 4.2.2 for now. Honestly had a lot of other work to do so I figured to postpone the update for now. I'll give 4.7.2 a try for our next release!

philips77, 2020-05-13:
Sure! But remember, that 4.7.2 is so much better than 4.2.2. It's better by 0.5.0 ;)

philips77, 2020-05-13:
Also, we switched from Zip to ZipFramework dependency at some point. Just FYI.

closed 2020-05-13 by philips77

AlexDM0, 2020-05-13:
I can't wait to see the massive boost of the 0.5 performance! Thanks for the help!