nordicsemi/IOS-DFU-Library #373: DFUServiceInitiator peripheral selector never called.

rodmaz, 2020-05-17:
**DFU Bootloader version (please complete the following information):**
 - SDK version: [e.g. SDK 16]
 - Bonding used: no
 - Library version: 4.7.2

**Device information (please complete the following information):**
 - Device :iPad mini 4 (Model A1538)
 - OS: iOS 13.4.1 (17E262)

**Describe the bug**
Using `DFUServiceInitiator` never calls our own `peripheralSelector` delegate. The class also does not call the default `DFUPeripheralSelector` class, if we use it.

Also there seems to be a confusing or poorly documented approach by the library regarding the `peripheralSelector`. 

**We are setting the device into DFU mode outside this library. The goal is to use the library to actually update the firmware.**

The goal of `peripheralSelector` is to locate (via delegate) the peripheral, whose firmware will be updated.

**Question:** why does `DFUServiceInitiator.start()` methods requires either the `CBPeripheral` or its UUID? 

Since our peripheral is already in DFU mode (broadcasting service `0xFE59`), **we expect to use the delegate to locate the peripheral**. However the delegate is never called. We actually do not expect to set its `CBPeriphera`l or its UUID in the first place in `start()`, but that is what the library also requires. 
**In the code below we use the device UUID before its goes into DFU mode.** This obviously does not work as the library attempts to connect to it apparently and times out:

```
2020-05-17 20:08:53.350921-0300 MyPeripheral[7090:3355926] initializing service initiator
2020-05-17 20:08:53.353290-0300 MyPeripheral[7090:3356444] starting service initiator
2020-05-17 20:08:53.354360-0300 MyPeripheral[7090:3356443] started DFU state: Connecting
2020-05-17 20:08:53.368644-0300 MyPeripheral[7090:3355926] [Callback] Central Manager did update state to: Powered ON
2020-05-17 20:08:53.369337-0300 MyPeripheral[7090:3355926] Connecting to DEVICE-NAME...
2020-05-17 20:08:53.369601-0300 MyPeripheral[7090:3355926] centralManager.connect(peripheral, options: nil)
2020-05-17 20:09:03.370046-0300 MyPeripheral[7090:3356441] centralManager.cancelPeripheralConnection(peripheral)
2020-05-17 20:09:03.377744-0300 MyPeripheral[7090:3356443] [Callback] Central Manager failed to connect to peripheral (timeout)
2020-05-17 20:09:03.385767-0300 MyPeripheral[7090:3356443] an error occurred while performing DFU firmware update in peripheral; Device failed to connect; DFUError
2020-05-17 20:09:03.389904-0300 MyPeripheral[7090:3356441] [AWLProvisioningStateMachine] an error occurred while updating firmware in DFU mode; errorWhilePerformingDFU
2020-05-17 20:09:03.391517-0300 MyPeripheral[7090:3356441] [AWLProvisioningStateMachine] AWLFirmwareUpdateState  -> AWLFinishedState
```

**Code**

```
var result: Result<Void, AWLBLEError>?
DispatchQueue.global(qos: .utility).sync {
    let semaphore = DispatchSemaphore(value: 0)
    self.onDFUFirmwareUpdateHandler = { error in
        if error != nil {
            result = .failure(.errorWhilePerformingDFU)
        } else {
            result = .success(())
        }
        semaphore.signal()
    }
    // Perform firmware update.
    DDLogDebug("initializing service initiator")
    let initiator = DFUServiceInitiator(queue: self.peripheralOperationsQueue, delegateQueue: self.peripheralOperationsQueue, progressQueue: self.peripheralOperationsQueue, loggerQueue: self.peripheralOperationsQueue).with(firmware: firmware)
    initiator.logger = self
    initiator.delegate = self
    initiator.peripheralSelector = self
    initiator.alternativeAdvertisingNameEnabled = false
    DDLogDebug("starting service initiator")
    let dfuController: DFUServiceController? = initiator.start(targetWithIdentifier: self.peripheralUUID)
    // Wait for firmware update.
    guard semaphore.wait(timeout: .now() + Constants.dfuFirmwareTimeout) == .success else {
        // Timeout occurred while performing firmware update.
        DDLogError("timeout error occurred while updating firmware")
        _ = dfuController?.abort()
        result = .failure(.errorWhilePerformingDFU)
        return
    }
}
```

And our delegate (**which is never called**) looks like:

```
extension AWLBLEWorker: DFUPeripheralSelectorDelegate {
    func select(_ peripheral: CBPeripheral, advertisementData: [String : AnyObject], RSSI: NSNumber, hint name: String?) -> Bool {
        DDLogVerbose("checking peripheral for DFU: \(peripheral)")
        return true
    }
    
    func filterBy(hint dfuServiceUUID: CBUUID) -> [CBUUID]? {
        DDLogVerbose("checking service UUID in DFU: \(dfuServiceUUID)")
        return [NordicDFU.serviceId]
    }
}
```

philips77, 2020-05-18:
The peripheral selector is used only in case when it is this library what switches the device into dfu mode. The device then starts advertising with a different address, and the lib is scanning for it and matching using the provided or default selector.

In your case, you need to scan for the device on your own. Remember, that in Secure Dfu the device in dfu bootloader mode uses a different Mac address, so it's not the same device (and not the same peripheral uuid) as you use for normal operations.

rodmaz, 2020-05-24:
@philips77 Thanks! I would recommend this aspect to be better documented though.

closed 2020-05-24 by rodmaz