nordicsemi/IOS-DFU-Library #375: Crash on SecureDFUPeripheral.swift, line 213

papfonseca, 2020-07-17:
**DFU Bootloader version (please complete the following information):**
 - SDK version: 12
 - Library version: 4.7.2

**Device information (please complete the following information):**
 - Device: iPhone 6S, XR
 - OS: 13.5.1

**Your question**
We have a retry logic on our app. When a progress is interrupted, we show an error to the user with a retry button. This retry logic is always crashing.

The flow is:
- Start an update
- Force a disconnect (walk away with device, for example). We'll get the error "Error: Device failed to connect, Error Type: 201"
- Retry the update with device near your phone
- The app will show the error "Error: Operation not permitted, Error Type: 18" and then crashes. The crash is the following.

`Unexpectedly found nil while unwrapping an Optional value: file /Pods/iOSDFULibrary/iOSDFULibrary/Classes/Implementation/SecureDFU/Peripheral/SecureDFUPeripheral.swift, line 213`

The `dfuService` is nil.

Can you help me on this?

Thanks!

papfonseca, 2020-07-17:
If it helps:

```
2020-07-17 17:47:27.330> [DFU] --> Connecting to *...
2020-07-17 17:47:27.330> [DFU] --> centralManager.connect(peripheral, options: nil)
2020-07-17 17:47:27.378> [DFU] --> [Callback] Central Manager did connect peripheral
2020-07-17 17:47:27.378> [DFU] --> Connected to *
2020-07-17 17:47:27.378> [DFU] --> Discovering services...
2020-07-17 17:47:27.378> [DFU] --> peripheral.discoverServices(nil)
2020-07-17 17:47:27.526> [DFU] --> Services discovered
2020-07-17 17:47:27.526> [DFU] --> Starting Secure DFU...
2020-07-17 17:47:27.527> [DFU] --> Connected to *
2020-07-17 17:47:27.527> [DFU] --> Services discovered
2020-07-17 17:47:27.527> [DFU] --> Secure DFU Service found
2020-07-17 17:47:27.527> [DFU] --> Discovering characteristics in DFU Service...
2020-07-17 17:47:27.527> [DFU] --> peripheral.discoverCharacteristics(nil, for: FE59)
2020-07-17 17:47:27.530> [DFU] --> DFU characteristics discovered
2020-07-17 17:47:27.534> [DFU] --> Enabling notifications for 8EC90001-F315-4F60-9FB8-838830DAEA50...
2020-07-17 17:47:27.534> [DFU] --> peripheral.setNotifyValue(true, for: 8EC90001-F315-4F60-9FB8-838830DAEA50)
2020-07-17 17:47:27.646> [DFU] --> Notifications enabled for 8EC90001-F315-4F60-9FB8-838830DAEA50
2020-07-17 17:47:27.646> [DFU] --> Secure DFU Control Point notifications enabled
2020-07-17 17:47:27.646> [DFU] --> Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2020-07-17 17:47:27.647> [DFU] --> peripheral.writeValue(0x0601, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2020-07-17 17:47:27.737> [DFU] --> Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2020-07-17 17:47:27.739> [DFU] --> Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600601000100008d000000bb32ba17
2020-07-17 17:47:27.739> [DFU] --> Command object info (Max size = 256, Offset = 141, CRC = 17BA32BB) received
2020-07-17 17:47:27.740> [DFU] --> Received CRC match Init packet
2020-07-17 17:47:27.741> [DFU] --> Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2020-07-17 17:47:27.741> [DFU] --> peripheral.writeValue(0x04, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2020-07-17 17:47:27.827> [DFU] --> Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2020-07-17 17:47:27.827> [DFU] --> Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600401
2020-07-17 17:47:27.828> [DFU] --> Command object executed
2020-07-17 17:47:27.829> [DFU] --> Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2020-07-17 17:47:27.829> [DFU] --> peripheral.writeValue(0x020c00, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2020-07-17 17:47:28.008> [DFU] --> Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2020-07-17 17:47:28.008> [DFU] --> Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600201
2020-07-17 17:47:28.008> [DFU] --> Packet Receipt Notif enabled (Op Code = 2, Value = 12)
2020-07-17 17:47:28.008> [DFU] --> Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2020-07-17 17:47:28.008> [DFU] --> peripheral.writeValue(0x0602, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2020-07-17 17:47:28.008> [DFU] --> Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2020-07-17 17:47:28.008> [DFU] --> Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600601001000005c030000c8574b39
2020-07-17 17:47:28.008> [DFU] --> Data object info (Max size = 4096, Offset = 860, CRC = 394B57C8) received
2020-07-17 17:47:28.008> [DFU] --> CRC does not match! Retrying...
8.008> [DFU] --> Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2020-07-17 17:47:28.014> [DFU] --> peripheral.writeValue(0x010200100000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2020-07-17 17:47:28.014> [DFU] --> Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
2020-07-17 17:47:28.014> [DFU] --> Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600101
2020-07-17 17:47:28.014> [DFU] --> Data object 1/52 created
2020-07-17 17:47:28.014> [DFU] --> wait(400)
2020-07-17 17:47:28.014> [DFU] --> Connected to *
2020-07-17 17:47:28.014> [DFU] --> Services discovered
2020-07-17 17:47:28.014> [DFU] --> Starting Secure DFU...
2020-07-17 17:47:28.014> [DFU] --> Connected to *
2020-07-17 17:47:28.014> [DFU] --> Services discovered
2020-07-17 17:47:28.014> [DFU] --> Secure DFU Service found
4> [DFU] --> Discovering characteristics in DFU Service...
2020-07-17 17:47:28.015> [DFU] --> peripheral.discoverCharacteristics(nil, for: FE59)
2020-07-17 17:47:28.016> [DFU] --> DFU characteristics discovered
2020-07-17 17:47:28.018> [DFU] --> Enabling notifications for 8EC90001-F315-4F60-9FB8-838830DAEA50...
2020-07-17 17:47:28.018> [DFU] --> peripheral.setNotifyValue(true, for: 8EC90001-F315-4F60-9FB8-838830DAEA50)
2020-07-17 17:47:28.155> [DFU] --> Notifications enabled for 8EC90001-F315-4F60-9FB8-838830DAEA50
2020-07-17 17:47:28.156> [DFU] --> Secure DFU Control Point notifications enabled
2020-07-17 17:47:28.156> [DFU] --> Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
2020-07-17 17:47:28.156> [DFU] --> peripheral.writeValue(0x0601, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
2020-07-17 17:47:28.216> [DFU] --> Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
[... cut ...]
Fatal error: Unexpectedly found nil while unwrapping an Optional value: file /Users/apple/Documents/*-ios/Pods/iOSDFULibrary/iOSDFULibrary/Classes/Implementation/SecureDFU/Peripheral/SecureDFUPeripheral.swift, line 213
2020-07-17 17:47:28.871644+0100 * Dev[5103:2268809] Fatal error: Unexpectedly found nil while unwrapping an Optional value: file /Users/apple/Documents/*-ios/Pods/iOSDFULibrary/iOSDFULibrary/Classes/Implementation/SecureDFU/Peripheral/SecureDFUPeripheral.swift, line 213
(lldb) 
```

philips77, 2020-07-17:
Thank you for the log and bug report. I'll look into it when I'm back at work.

philips77, 2020-07-31:
Hello,
Why are you starting another DFU while old one is still in progress:
```
2020-07-17 17:47:28.014> [DFU] --> Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600101
2020-07-17 17:47:28.014> [DFU] --> Data object 1/52 created
2020-07-17 17:47:28.014> [DFU] --> wait(400)
2020-07-17 17:47:28.014> [DFU] --> Connected to * 
```
And here come lines from here come from another DFU process, but the old one didn't complete nor fail.
```
2020-07-17 17:47:28.014> [DFU] --> Services discovered
2020-07-17 17:47:28.014> [DFU] --> Starting Secure DFU...
[...]
```
And then both processes interrupt each other.

philips77, 2020-07-31:
The error was caused by the first dfu process that finished and started cleaning up, i.e. it nullified the `dfuService` instance.
At this point the 2nd progress (or more?) is in a very weird state, where there's nothing really that can be done. One could argue that I could not force-unwrap those variables, but in my opinion thanks to that you know you have something wrong. If it would just quietly stop updating leaving some unwanted resources you wouldn't find out about it.

papfonseca, 2020-08-02:
I really appreciate your help. I'll check it again. Thank you!

closed 2021-04-29 by philips77