nordicsemi/IOS-DFU-Library #376: Stuck on Connecting

canedo2, 2020-07-31:
**DFU Bootloader version (please complete the following information):**
 - SDK version: 15.3.0
 - Bonding used: NO
 - Library version: 4.7.2

**Device information:**
 - Device: iPhone X
 - OS: iOS 13.1

**Your question**
Hello guys, 
We are having an issue with the OTA update on our BLE device in iOS only. We were only able to update our device once with our app, although we are not having problems with the nRF Connect app. 

We got it once in a device:
[image]

The rest of our tries result on state getting stuck on Connecting (delegate reports this step only):
[image]

We are unable to get any log to have an idea on what's happening. Maybe you can help us to fix this.

Thank you guys.

**Code**
`class OtaUpdateViewController: StatefulViewController<HasBandStore & HasBluetoothStore, OtaUpdateNavigator> & MenuDetailViewController, DFUServiceDelegate, DFUProgressDelegate, LoggerDelegate {
        @IBOutlet var update1Button: UIButton!
        @IBOutlet var update4Button: UIButton!
        @IBOutlet var progressView: UIProgressView!
        @IBOutlet var speedLabel: UILabel!
        @IBOutlet var statusLabel: UILabel!
        @IBOutlet var textView: UITextView!
        override func viewDidLoad() {
            super.viewDidLoad()
            setupRx()
            setupLiterals()
        }

        override func viewWillAppear(_ animated: Bool) {
            super.viewWillAppear(animated)
            set(title: Literals.OptionBandTitle)
            applyStyle()
        }

        private func setupRx() {}

        private func setupLiterals() {}

        private func applyStyle() {
            update1Button.style(.blueTextBoldLightBlueBackgroundButton)
            update4Button.style(.blueTextBoldLightBlueBackgroundButton)
        }

        // MARK: - functions

        fileprivate func update(withFileURL url: URL?) {
            if let url: URL = url, let selectedFirmware = DFUFirmware(urlToZipFile: url) {
                BluetoothService.sharedInstance.connectedPeripheral.do(onNext: { peripheral in
                    let initiator = DFUServiceInitiator().with(firmware: selectedFirmware)
                    initiator.logger = self // - to get log info
                    initiator.delegate = self // - to be informed about current state and errors
                    initiator.progressDelegate = self // - to show progress bar
                    initiator.alternativeAdvertisingNameEnabled = false
                    let controller = initiator.start(target: peripheral.peripheral)

                }).subscribe().disposed(by: disposeBag)
            }
        }

        // MARK: - Actions

        @IBAction func updateLeftActionTap(_: UIButton) {
            textView.text = "Updating to ota_v1.1.4\n"
            let urlPath = Bundle.main.url(forResource: "ota_v1.1.4", withExtension: "zip")
            update(withFileURL: urlPath)
        }

        @IBAction func updateRightActionTap(_: Any) {
            textView.text = "Updating to ota_v1.2.0\n"
            let urlPath = Bundle.main.url(forResource: "ota_v1.2.0", withExtension: "zip")
            update(withFileURL: urlPath)
        }

        // MARK: - Nordic delegates

        func dfuStateDidChange(to state: DFUState) {
            statusLabel.text = state.description()
        }

        func dfuError(_: DFUError, didOccurWithMessage message: String) {
            let alert = UIAlertController(title: "Error", message: message, preferredStyle: .alert)
            let okAction = UIAlertAction(title: "OK", style: .default, handler: nil)
            alert.addAction(okAction)
            present(alert, animated: true, completion: nil)
        }

        func dfuProgressDidChange(for part: Int, outOf totalParts: Int, to progress: Int, currentSpeedBytesPerSecond: Double, avgSpeedBytesPerSecond _: Double) {
            speedLabel.text = "\((currentSpeedBytesPerSecond / 1_024.0).rounded(toPlaces: 2)) byte/s"
            let progressPercent = Float(progress) / 100.0
            progressView.progress = progressPercent
        }

        func logWith(_ level: LogLevel, message: String) {
            if let position = textView.textRange(from: textView.beginningOfDocument, to: textView.beginningOfDocument) {
                textView.replace(position, withText: "\(level.name()) - \(message)\n ")
            }
        }
    }
}`

philips77, 2020-08-03:
After the update, does your device advertise as usual? Can you trigger the bootloader mode? Can you do DFU multiple times using nRF Connect (Android or iOS)?

canedo2, 2020-08-03:
Hey!

Yes, the device advertises as usual
Yes (tested with nRF Connect for iOS)
Yes, tested in Android

Thanks for your help!

philips77, 2020-08-04:
We've seen an issue (https://devzone.nordicsemi.com/f/nordic-q-a/64010/ios-devices-suddenly-can-no-longer-connect-or-discover-services) where iOS can no longe connect to a device, while Android can. Perhaps it's similar? Did you try other app, like nRF Toolbox for iOS?

canedo2, 2020-08-12:
Hey @philips77! 
It seems to be working now, we didn't anything especial, maybe it was problem in our hardware. I'm closing this by now.
Anyway, thank you for your help and time.

closed 2020-08-12 by canedo2