nordicsemi/IOS-DFU-Library #380: Writing to characteristic failed. Check if Service Changed service is enabled

A-i-r-n, 2020-08-06:
**Information**
This GitHub Issue page is for reporting issues or asking questions regarding the iOS DFU library. For general DFU questions, SDK questions, etc, please check our [DevZone](https://devzone.nordicsemi.com/).
Make sure you are using the latest version of the library: [image]
Also, before asking a question, make sure similar issue isn't already opened in [open](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues) or [closed](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues?q=is%3Aissue+is%3Aclosed) issues.

**DFU Bootloader version (please complete the following information):**
 - SDK version: [e.g. SDK 16]
 - Bonding used: [yes/no]
 - Library version: [e.g. 4.6.1]

**Device information (please complete the following information):**
 - Device: [e.g. iPhone 11]
 - OS: [e.g. iOS 13.1]

**Your question**
What do you have problem with?

**Logs**
Add any other context or logs describing the problem here.

philips77, 2020-08-06:
Please provide the above information in the issue, non in the issue name.

philips77, 2021-04-29:
This may be related to #395. As no additional info was provided, I'm closing the issue.

closed 2021-04-29 by philips77