nordicsemi/IOS-DFU-Library #381: Fatal error: Unexpectedly found nil while unwrapping an Optional value: file

ronhuiyuan, 2020-08-11:
Pods/iOSDFULibrary/iOSDFULibrary/Classes/Implementation/LegacyDFU/Services/LegacyDFUService.swift, line 292

Fatal error: Unexpectedly found nil while unwrapping an Optional value: file

Can you help me

ronhuiyuan, 2020-08-12:
File Name: AI FB_4.2.60_20200731_alpha.zip
Parts: 1
Size: 49 KB
Soft Device Size: Zero KB
Bootloader Size: Zero KB
[Callback] Central Manager did update state to: Powered ON
Connecting to AI FB...
centralManager.connect(peripheral, options: nil)
[Callback] Central Manager did connect peripheral
Connected to AI FB
Discovering services...
peripheral.discoverServices(nil)
Services discovered
Starting Legacy DFU...
Connected to AI FB
Services discovered
Legacy DFU Service found
Discovering characteristics in DFU Service...
peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
DFU characteristics discovered
Reading DFU Version number...
peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0100
Version number read: 0.1
Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
DFU Control Point notifications enabled
Application with buttonless update found
Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
Data written to 00001531-1212-EFDE-1523-785FEABCD123
Jump to bootloader (Op Code = 1, Upload Mode = 4) request sent
[Callback] Central Manager did disconnect peripheral
Disconnected by the remote device
Connecting to AI FB...
centralManager.connect(peripheral, options: nil)
[Callback] Central Manager did connect peripheral
Connected to AI FB
Discovering services...
peripheral.discoverServices([00001530-1212-EFDE-1523-785FEABCD123])
Services discovered
Legacy DFU Service found
Discovering characteristics in DFU Service...
peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
DFU characteristics discovered
DFU Control Point characteristic not found
Disconnecting...
centralManager.cancelPeripheralConnection(peripheral)
[Callback] Central Manager did disconnect peripheral
Disconnected

StanislasGaitUp, 2020-08-18:
Hi, I found out that when I am requesting a DFU when the sensor is already connected, there is a crash in the library
[image]

Here is the code I am using to do the dfu
`
let dfuInitiator = DFUServiceInitiator(queue: DispatchQueue(label: "Other"))
dfuInitiator.delegate = self
dfuInitiator.progressDelegate = self
dfuInitiator.alternativeAdvertisingNameEnabled = false

if #available(iOS 11.0, macOS 10.13, *) {
    dfuInitiator.packetReceiptNotificationParameter = 0
}
_ = dfuInitiator.with(firmware: AppState.firmware).start(target: core)

`

philips77, 2020-08-18:
Thank you. I'll look into that.

StanislasGaitUp, 2020-08-19:
Thanks !
To work around the problem I am now disconnecting from the sensor before starting the update, waiting 1 second, do the update, and reconnect

philips77, 2021-04-29:
Force unwrapping has now been fixed everywhere, so this issue should not appear again. Change will be released in 4.10.

closed 2021-04-29 by philips77