nordicsemi/IOS-DFU-Library #383: Invalid Bundle error after uploading build to AppStore

tomaszoraczewski-silvair, 2020-08-27:
- Library version: 4.8.0
- Carthage: 0.34
- Xcode: 11.6
- Fastlane: 2.156.1

After updating the library to 4.8.0 from 4.6.1, the build that is published to AppStore by Fastlane is invalid. Downgrading to 4.6.1 resolves it. The library is installed by Carthage. App Store Connect sends an email with the following error:
> ITMS-90562: Invalid Bundle - The app submission can not be successfully recompiled from bitcode due to missing symbols during linking. You can try to reproduce and diagnose such issues locally by following the instructions from: https://developer.apple.com/library/archive/technotes/tn2432/_index.html

Note:
`Enable Bitcode` setting in Xcode Build Settings is set to `YES`

philips77, 2020-08-27:
We had some issues reported, that 4.8.0 causes problems and crashes, but clearing Xcode cache fixes it. Did you try to set up Carthage dependency again?
Checking the releases, only 4.7.2 touched anything related to Carthage: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/releases

philips77, 2020-08-27:
Also, 4.7.0 migrated to Swift 5.2.

tomaszoraczewski-silvair, 2020-08-27:
I've tried to set up it again completely from scratch (removing all Carthage caches etc.) and it didn't help. It could also be somehow connected with Fastlane and Xcode itself, however, I submitted the issue here as the direct reason for it is changing DFU library version.

philips77, 2020-09-30:
Hi, sorry for the long delay. Did you manage to solve the issue?

tomaszoraczewski-silvair, 2020-09-30:
No, I'm still at 4.6.1. I don't have a direct need to go with a newer version.

marekniedbach-silvair, 2021-01-12:
It looks like the library in version 4.9.0 is working ok 👍

closed 2021-01-20 by philips77