nordicsemi/IOS-DFU-Library #384: EXC_BREAKPOINT

guopeng12, 2020-10-01:
Hello there. The version I use is iOSDFULibrary (4.8.0), and the test is no problem, but the online capture log shows that some users have reported errors.
code:
  DFUFirmware *firmware = [[DFUFirmware alloc] initWithUrlToZipFile:file];// or
     DFUServiceInitiator *initiator =[ [DFUServiceInitiator alloc]initWithQueue:dispatch_get_main_queue() delegateQueue:dispatch_get_main_queue() progressQueue:dispatch_get_main_queue() loggerQueue:dispatch_get_main_queue()];
     initiator = [initiator withFirmware:firmware];
     initiator.progressDelegate = self; //-to show progress bar
     [initiator startWithTarget:self.cbPeripheral];
Error log:
    EXC_BREAKPOINT 0x00000001cda4c7dc
Number of error lines:
    [initiator startWithTarget:self.cbPeripheral];

thank you very much

philips77, 2020-10-02:
This seems to be an error with caching, I think. We also had this issue, but clearing Xcode cache fixed it. As if iOS was trying to call old method, not from updated library. Don't know, however, how can this happen in production when you don't have it and you recompiled the app.

guopeng12, 2020-10-02:
Thank you very much for your reply. I did not understand the last sentence. Do you mean to clear the xcode cache, repackage and upload a build?

philips77, 2020-10-02:
In my opinion after you fix this locally, it should also compile properly and work on all clients' phones. I don't know why it may fail on some, but not others. As I understand this issue it's a "Null pointer", where the app tries to call invalid address, as the lib wasn't compiled properly. After all, clearing Derived Data and caches fixes it locally. 

Perhaps @Dinesharjani or @NickKibish can help here?

Btw, I released 4.9.0 just now. Check out #274.

guopeng12, 2020-10-09:
Thank you very much for your reply. I have upgraded to 4.9.0. We will submit another version of our app later to see the effect.

closed 2021-04-29 by philips77