nordicsemi/IOS-DFU-Library #390: Xcode 12 and maybe new requiriment

Lucasark, 2020-11-07:
**Information**
This GitHub Issue page is for reporting issues or asking questions regarding the iOS DFU library. For general DFU questions, SDK questions, etc, please check our [DevZone](https://devzone.nordicsemi.com/).
Make sure you are using the latest version of the library: [image]
Also, before creating a new issue, make sure similar issue isn't already opened in [open](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues) or [closed](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues?q=is%3Aissue+is%3Aclosed) issues.

**DFU Bootloader version (please complete the following information):**
 - SDK version: [e.g. SDK 16]
 - Bonding used: [yes/no]
 - Library version: 4.9.0

**Device information (please complete the following information):**
 - Device: iPhone 12
 - OS: iOS 13

**Describe the bug**
A clear and concise description of what the bug is.

After upgrade my Xcode to 12.1 (not Beta) I got this error:

'DFUFirmware' with definition in module 'iOSDFULibrary.Swift' has different definitions in different modules; first difference is this method
'DFUFirmwareSize' has different definitions in different modules; first difference is definition in module 'iOSDFULibrary.Swift' found 3
'DFUServiceController' with definition in module 'iOSDFULibrary.Swift' has different definitions in different modules; first difference is this method
'DFUServiceInitiator' has different definitions in different modules; first difference is definition in module 'iOSDFULibrary.Swift' found property name 'forceScanningForNewAddressInLegacyDfu'

I already update my pod to 4.9.0 to catch last PR that mention Xcode 12. 

[image]

Temporary Workaround:
Downgrade to 4.8.0 and using Xcode 11.7

In this link mention that:
"This problem occurs after XCode 12 Beta5. Xcode doesn't allow different modules to define same names (Probably for public classes & protocols). Alamofire and Kingfisher appears to define SessionDelegate at the same time. I'm still trying to find a solution.."

https://stackoverflow.com/questions/63614048/xcode-12-sessiondelegate-has-different-definitions-in-different-modules

**Logs**
Add any other context or logs describing the problem here.

[image]

octalanil1, 2020-12-09:
[image]
[image]

I'm also facing same error in xcode 12.1.
If you find any solution please let me know ASAP.

jsimonassi, 2020-12-09:
This problem happened to me because iOSDFULibrary had the same class / method name as IDOBluetooth (A .framework also used in my project). I managed to build with a few steps:

1 - Search for the conflicting class / method by filtering through the iOSDFULibrary folder inside your Pods folder;
[image]

2 - Rename all occurrences. (In my case, I renamed DFUFirmware to DFUFirmwareBle);

3 - Try to build. At this point I received this error:
[image]

I solved it with a cmd + k.

4 - Rename all calls within your project that use the iOSDFUFirmware methods / classes. XCode will help you with this part by showing the errors.

At this point, the build was done successfully for me.

octalanil1, 2020-12-10:
No it's not working.
If saw screenshoot it's clearly showing my problem is in DFUServiceInitiator class and DFUServiceDelegate.
So any othe solution please let me know asap, i stuck in this pb from last two days.

Lucasark, 2020-12-10:
> No it's not working.
> If saw screenshoot it's clearly showing my problem is in DFUServiceInitiator class and DFUServiceDelegate.
> So any othe solution please let me know asap, i stuck in this pb from last two days.

@octalanil1 My problem is not same is yours. 
Your error is more like to using same name of class in differents files of your project.
I recommend you print your log error. Or, maybe ready carefully.

I work with @jsimonassi t solve this.  And, have sucess. So, i am closing this issue.
But, make a alert for who is using more than one SDK that use iOSDFU.

closed 2020-12-10 by Lucasark