nordicsemi/IOS-DFU-Library #391: After firmware update application can't detect peripherals

ltAldoRaine, 2020-11-10:
Hello, I'm using this library for updating firmware to my device. Update process goes well, problem occurs when it finishes application can't detect Bluetooth peripherals. Do you have any suggestions ?

9Emil, 2020-11-19:
I have had the same issue.
My issue is that the device connects, but the phone can't see the device.

Currently I have found a workaround, either restart your phone or in settings turn bluetooth off and on again. (Does not work if not done from settings)

ltAldoRaine, 2020-11-19:
I'm restarting application, only that helps me.

philips77, 2020-11-19:
Hello,
Sorry, I'm on a leave until mid January. I'll look into that when I'm back at the office.

bigbearlazier, 2020-11-23:
Try making your device send a Service Changed indication after updating its firmware. It may help.

philips77, 2021-04-29:
Hi,
Did you manage to solve the problem? Can we close this issue?

philips77, 2021-07-30:
Closing due to inactivity. Please, reopen if needed.

closed 2021-07-30 by philips77