nordicsemi/IOS-DFU-Library #393: SPM not working on Xcode 12.2

hohteri, 2020-11-17:
I don’t know why but NordicDFU (4.9.0) doesn’t get version number and isn’t available in Xcode 12.2 SPM. 

It will show up in Xcode but without version number and isn’t compiled thus failing a build depending on it. 

And apologies for badly formatted issue report.

philips77, 2020-11-18:
Thank you. We'll look into that.

dinesharjani, 2020-11-19:
Hi @hohteri ,

Can you provide some output from Xcode regarding the specific issue? We were able to add NordicDFU package from scratch into an iOS project, and Xcode 12.2's SPM correctly read the version as 4.9.0 and everything appears to be working (building & running at least).

We suggest, if you haven't done it already, to remove the SPM dependency/framework and re-adding it again.

hohteri, 2020-11-19:
Works today, thanks for looking into this. Closing!

closed 2020-11-19 by hohteri