nordicsemi/IOS-DFU-Library #398: DFU Progress not being reported and state doesn't transition to completed in release mode

pdecostervgls, 2021-01-20:
**DFU Bootloader version (please complete the following information):**
 - SDK version: 16
 - Bonding used: no
 - Library version: 4.9.0
 - Binding Library: https://github.com/Laerdal/Laerdal.Xamarin.Dfu.iOS

**Device information (please complete the following information):**
 - Device: iPhone 11
 - OS: iOS 14.3

**Describe the bug**
When the application runs in release mode, the DFUProgressDelegate does not get called and the DFU State also doesn't change to completed. The process does however complete and the device gets rebooted.

**Logs**
Because the application is compiled in release mode, the amount of logs I can provide is very limited. If anyone has an idea how to get logs in release mode,  I will be happy to provide them.

tompi, 2021-01-20:
@pdecostervgls If this works fine in debug mode, Im guessing its something with your linker settings. Check if it works with "Dont link", if it does, change it back(link everythin) and add exceptions for your classes to your linkdescription until it works...

You probably should create ticket in the laerdal repository, since your using "our" wrapper for this lib :)

pdecostervgls, 2021-01-20:
@tompi Setting the release mode to "don't link" doesn't help, but probably because other assemblies that I skipped with `--linkskip` don't get skipped anymore. Although that seems weird because I would assume "Don't link" would make it work anyway...

```Xamarin.iOS: Unable to locate assembly 'Microsoft.EntityFrameworkCore.resources' (culture: 'nl-NL')```
This doesn't seem related to this issue.

tompi, 2021-01-21:
@pdecostervgls  I think you misunderstood what the "--linkskip" parameter does. It basically tells the linker which assemblies to skip. You can ALSO do that in the "linkdescription" file, which I think is better(more transparent). If you say "dont link" the linker will not run at all, so everything is included, no matter what you specified in "--linkskip"(so "linkskip" actually ensures things are INCLUDED).

Anyway, perhaps if you provide your code to instantiate and attach the processdelegate and the actual "start" call, it could be easier to read. (if you make your own processdelegate it would also be nice to see this...)

pdecostervgls, 2021-01-21:
@tompi I can show the C# code that I'm using, thats no problem.

```
...
// Create and register delegates
iOSDFUServiceDelegate dfuDelegate = new iOSDFUServiceDelegate();
dfuDelegate.onStateChanged += DfuDelegate_onStateChanged;

dfuSI.Delegate = dfuDelegate;
dfuSI.Logger = new iosLogger();

iOSProgressDelegate progressDelegate = new iOSProgressDelegate();
progressDelegate.onProgressChanged += ProgressDelegate_onProgressChanged;
dfuSI.ProgressDelegate = progressDelegate;

// Set Firmware Package
dfuSI.WithFirmware(dfuFirmware);

// Start DFU Process
DFUServiceController controller = dfuSI.Start();
...
private void ProgressDelegate_onProgressChanged(object sender, int e)
{
    this.onProgressChanged(e);
}

private void DfuDelegate_onStateChanged(object sender, DFUState e)
{
    this.dfuState = e;
}
...
public class iosLogger : LoggerDelegate, ILoggerDelegate
{
    public override void Message(LogLevel level, string message)
    {
        System.Diagnostics.Debug.WriteLine(message);
    }
}
    
public class iOSProgressDelegate : DFUProgressDelegate
{
    public event EventHandler<int> onProgressChanged;
    public override void OutOf(nint part, nint totalParts, nint progress, double currentSpeedBytesPerSecond, double avgSpeedBytesPerSecond)
    {
        this.onProgressChanged?.Invoke(null, (int)progress);
    }
}

public class iOSDFUServiceDelegate : DFUServiceDelegate
{
    public event EventHandler<DFUState> onStateChanged;
    public override void DfuError(DFUError error, string message)
    {
        
    }

    public override void DfuStateDidChangeTo(DFUState state)
    {
        this.onStateChanged?.Invoke(this, state);
    }
}
```

As for the threading, I am logging the callbacks to the console, I also have the debugging switch on to get extensive logging from the DFU library.

tompi, 2021-01-21:
Strange... maybe @framinosona has any tips on what to try?

framinosona, 2021-01-21:
Oh Yes sorry missed this bug, thanks @tompi ! 
I've seen this behavior before, it does ring a bell ... I agree with @tompi that if it works in debug and not in release it smells like a linking issue. 
@pdecostervgls which version of the NUGET are you using ? I've released version 4.9.0.13 that should have fixed some linking issues.

pdecostervgls, 2021-01-21:
@framinosona I'm using version 4.9.0.13

framinosona, 2021-01-21:
Okay, here are a few things you can try: 
- Using DFUServiceInitiator.WeakProgressDelegate and WeakDelegate instead of ProgressDelegate and Delegate
- Make sure your initiator uses the right Thread Queue, here is what I use:

```csharp
new DFUServiceInitiator(DispatchQueue.GetGlobalQueue(DispatchQueuePriority.Default),
                                                 DispatchQueue.GetGlobalQueue(DispatchQueuePriority.Default),
                                                 DispatchQueue.GetGlobalQueue(DispatchQueuePriority.Default),
                                                 DispatchQueue.GetGlobalQueue(DispatchQueuePriority.Default))
```

framinosona, 2021-01-21:
Ahh, you're using a deprecated constructor: 
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/d4a1da27fbf195b325cea3036235a4939a8f4efa/iOSDFULibrary/Classes/Implementation/DFUServiceInitiator.swift#L332

The latest versions of the iOS code creates its own manager, that helps prevent your manager and its delegates to be stripped apart. 
Then pass the identifier of your device : this.Controller = this.Initiator.WithFirmware(dfuPackage).StartWithTargetWithIdentifier(identifier);

pdecostervgls, 2021-01-21:
@framinosona I made some changes like you suggested, doesn't change the behavior.

pdecostervgls, 2021-01-22:
@tompi @framinosona First of all, Thank you both for the support so far, second of all, I have a confession to make. It turns out I wasn't properly logging the callbacks anymore. I used to have logging code in there, but it was removed at a later time apparently. When I re-added the logging code I noticed it being called with progress and state change messages. This caused me to experiment with the Queues that get passed to the ServiceInitiator, ultimately ending up with passing the `DispatchQueue.MainQueue` as arguments, for all delegates. This caused the progress event to originate from the right thread making my UI work.

However, I still don't get the stateChanged completed message

framinosona, 2021-01-22:
@pdecostervgls Thanks for your help making our package better :) 
It happens :) :) 
It was a year ago, but I remember putting DispatchQueue.GetGlobalQueue(DispatchQueuePriority.Default) because I was having issue with logging and state change too. I was updating multiple devices at once and on the less powerful iPhones I would only get the beginning of the feedback and then nothing, also seeing slowdowns on other tasks ... 

Maybe you should implement your DfuError callback? If an error occurs there is no state change anymore. I've added an error state to my state enum and send a statechanged : error

pdecostervgls, 2021-01-22:
@framinosona I added an Console.WriteLine in the DFU Error callback, but it doesn't trigger. Also, the DFU operation finishes successfully everytime, judging by the status of the device.

pdecostervgls, 2021-01-26:
So just a little update on my side:
Right now there are 3 scenarios:
* Sometimes I dont get progress notifications, also not being logged, but at the end I get a completed state and am able to finish the procedure.
* sometimes I do get progress notifications, but they stop before I reach the 100%, I also don't get a completed or disconnecting state. 
* sometimes I do get progress notifications, and they persist until 100%, but I dont get a completed or disconnecting state.

framinosona, 2021-01-26:
It's strange because I remember having that exact same behavior too but I don't remember how we fixed it ... But playing with the DispatchQueue helped. I see many old issues about this :
https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/issues/103
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/issues/156

Most of the time the reason I see is a disposed resource, the Central or the peripheral.

philips77, 2021-04-27:
Hello,
Did you manage to fix the issue? Can we close it here?

pdecostervgls, 2021-04-27:
@philips77 unfortunately not, I will tackle this issue again soon as its still remaining on my to-do list. Will keep you updated.

framinosona, 2021-04-27:
Opened new issue on our repo @philips77

closed 2021-04-28 by philips77