nordicsemi/IOS-DFU-Library #399: Error while creating ZIP stream: The operation couldn’t be completed. (iOSDFULibrary.DFUStreamZipError error 0.)

mikkomanoza, 2021-01-21:
**Information**
This GitHub Issue page is for reporting issues or asking questions regarding the iOS DFU library. For general DFU questions, SDK questions, etc, please check our [DevZone](https://devzone.nordicsemi.com/).
Make sure you are using the latest version of the library: [image]
Also, before creating a new issue, make sure similar issue isn't already opened in [open](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues) or [closed](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues?q=is%3Aissue+is%3Aclosed) issues.

**DFU Bootloader version (please complete the following information):**
 - SDK version: [e.g. SDK 16]
 - Bonding used: [yes/no]
 - Library version: [e.g. 4.6.1]

**Device information (please complete the following information):**
 - Device: [e.g. iPhone 11]
 - OS: [e.g. iOS 13.1]

**Describe the bug**
A clear and concise description of what the bug is.
"Error while creating ZIP stream: The operation couldn’t be completed.(iOSDFULibrary.DFUStreamZipError error 0.)"

**Logs**

Hi, i am new to this library. I follow and read the documentation, but i'm stuck because of this error -
"Error while creating ZIP stream: The operation couldn’t be completed.(iOSDFULibrary.DFUStreamZipError error 0.)"
how do I fix it? Thanks.

my code 
   
    import iOSDFULibrary

    var heartRatePeripheral: CBPeripheral!
    var dfuController: DFUServiceController!

    private func startUpdate(localFilePathURLStr: String) {

        let fileUrl = URL(fileURLWithPath: localFilePathURLStr)
        let selectedFirmware = DFUFirmware(urlToZipFile: fileUrl)
        guard let firmware = selectedFirmware else { return }

        let initiator = DFUServiceInitiator(queue: nil).with(firmware: firmware)

        initiator.logger = self
        initiator.delegate = self
        initiator.progressDelegate = self
        initiator.enableUnsafeExperimentalButtonlessServiceInSecureDfu = true;
        dfuController = initiator.start(target: heartRatePeripheral)
    }

philips77, 2021-01-21:
Did you create your ZIP using [nrf util](https://github.com/NordicSemiconductor/pc-nrfutil)?

mikkomanoza, 2021-01-25:
I did not create the zip file itself. how do I know if the zip file was created in nrf util? thanks @philips77

philips77, 2021-01-25:
Check if you have the manifest.json file inside and 2 or 4 files with names equal as in the manifest.

mikkomanoza, 2021-01-25:
the zip file I have is this (screenshot 1), but when I unzip the firmware.zip it contains manifest.json (screenshot 2), is firmware.zip the right one to use?, thank you for the quick answer @philips77 

screenshot 1: https://ibb.co/xj5ckNp
screenshot 2: https://ibb.co/19ttbcq

philips77, 2021-01-25:
This is the error you're getting:
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/d4a1da27fbf195b325cea3036235a4939a8f4efa/iOSDFULibrary/Classes/Utilities/Streams/DFUStreamZip.swift#L34-L52
Try printing the localized error message and see exactly what is the reason. Your zip file looks ok. You may also use the above mentioned nrfutil to read the content. Or debug where exactly you are getting that error from, which line.

closed 2021-02-01 by mikkomanoza

AbdElrahman-Rafaat-Amer, 2023-06-25:
Hi @philips77,  I have the same error. This is the localized error message: "The operation couldn’t be completed. (ZIPFoundation.Archive.ArchiveError error 0.)"

Do you have any solution?

philips77, 2023-06-26:
Hi, is your device based on nRF5 SDK, or nRF Connect SDK? In the latter case you should use nRF Connect Device Manager library instead.