nordicsemi/IOS-DFU-Library #404: Correct values for uuidHelper or how to set requiredServices

connyhald, 2021-02-17:
We are doing Secure Buttonless DFU with our own characteristic inside our own service.

```swift
    private let dfu_uuid = [
        DFUUuid(withUUID: CBUUID(string: ourServiceUUID), forType:.secureService),
        DFUUuid(withUUID: CBUUID(string: commandCharactUUID), forType:.buttonlessWithoutBondSharing)
    ]
```

The DFU process works up to the point where the peripheral gets rebooted into boot loader mode and we connect to the "new" peripheral called `DfuTargetCorp`.

Now the library tries to discover services but it does not discover all services, instead it uses `peripheral.discoverServices(requiredServices)`. The issue now is, that in our case `requiredServices == [ourServiceUUID]`. However once the peripheral is in boot loader mode, it should be `0xFE59` (the default DFU Service) instead.

So my question is: How can I change `requiredServices` while discovering services in boot loader mode, but keep it as is while in application mode?

**DFU Bootloader version (please complete the following information):**
 - SDK version: 15.3
 - Bonding used: no
 - Library version: 4.9.0

**Device information (please complete the following information):**
 - Device: iPad 8 gen
 - OS: iOS 13.4

**Logs**
```
15:06:56.508 ℹ️ INFO DfuActivity.logWith():36 - DFU LIB: Connected to DfuTargCorp
15:06:56.516 ℹ️ INFO DfuActivity.logWith():36 - DFU LIB: Discovering services...
15:06:56.517 ℹ️ INFO DfuActivity.logWith():36 - DFU LIB: peripheral.discoverServices([XXX-ourServiceUUID-from-above-XXX])
15:06:56.517 ℹ️ INFO DfuActivity.logWith():36 - DFU LIB: Services discovered
15:06:56.518 ℹ️ INFO DfuActivity.logWith():36 - DFU LIB: DFU Service not found
15:06:56.519 ℹ️ INFO DfuActivity.logWith():36 - DFU LIB: No services found
15:06:56.520 ℹ️ INFO DfuActivity.logWith():36 - DFU LIB: Did you connect to the correct target? It might be that the previous services were cached: toggle Bluetooth from iOS settings to clear cache. Also, ensure the device contains the Service Changed characteristic
15:06:56.521 ℹ️ INFO DfuActivity.logWith():36 - DFU LIB: Disconnecting...
15:06:56.522 ℹ️ INFO DfuActivity.logWith():36 - DFU LIB: centralManager.cancelPeripheralConnection(peripheral)
15:06:56.523 ℹ️ INFO DfuActivity.logWith():36 - DFU LIB: [Callback] Central Manager did disconnect peripheral
15:06:56.524 ℹ️ INFO DfuActivity.logWith():36 - DFU LIB: Disconnected

```

philips77, 2021-02-17:
Hello,
What if you set [`.buttonlessService`](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/d4a1da27fbf195b325cea3036235a4939a8f4efa/iOSDFULibrary/Classes/Implementation/DFUUuidHelper.swift#L51) to your custom service, leaving `.secureService` with the default value?
Perhaps you'd have to set this [flag](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/d4a1da27fbf195b325cea3036235a4939a8f4efa/iOSDFULibrary/Classes/Implementation/DFUServiceInitiator.swift#L293) to *true*. I don't know if this will work, but maybe?

philips77, 2021-02-17:
You can also trigger your device to DFU bootloader mode on your own, scan for the device in that mode and initiate DFU for the new found device, so no UUID overriding is necessary.

connyhald, 2021-02-18:
Thanks for you answsers!

> What if you set [`.buttonlessService`](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/d4a1da27fbf195b325cea3036235a4939a8f4efa/iOSDFULibrary/Classes/Implementation/DFUUuidHelper.swift#L51) to your custom service, leaving `.secureService` with the default value?

If I set it up this way, we don't even reach the reboot because the lib cannot find a DFU Service. 

```
10:39:16.268 ℹ️ INFO DfuActivity.logWith():35 - DFU LIB: Discovering services...
10:39:16.268 ℹ️ INFO DfuActivity.logWith():35 - DFU LIB: peripheral.discoverServices(nil)
10:39:16.269 ℹ️ INFO DfuActivity.logWith():35 - DFU LIB: Services discovered
10:39:16.269 ℹ️ INFO DfuActivity.logWith():35 - DFU LIB: DFU Service not found
10:39:16.269 ℹ️ INFO DfuActivity.logWith():35 - DFU LIB: The following services were discovered:
10:39:16.269 ℹ️ INFO DfuActivity.logWith():35 - DFU LIB:  - XXX-ourServiceUUID-from-above-XXX
10:39:16.270 ℹ️ INFO DfuActivity.logWith():35 - DFU LIB:  - 180A
10:39:16.270 ℹ️ INFO DfuActivity.logWith():35 - DFU LIB: Did you connect to the correct target? It might be that the previous services were cached: toggle Bluetooth from iOS settings to clear cache. Also, ensure the device contains the Service Changed characteristic
10:39:16.270 ℹ️ INFO DfuActivity.logWith():35 - DFU LIB: Disconnecting...
``` 

> Perhaps you'd have to set this [flag](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/d4a1da27fbf195b325cea3036235a4939a8f4efa/iOSDFULibrary/Classes/Implementation/DFUServiceInitiator.swift#L293) to _true_. I don't know if this will work, but maybe?

Yes, that in combination with the above seem to work. So now I have:

```swift
    private let dfu_uuid = [
        DFUUuid(withUUID: CBUUID(string: ourServiceUUID), forType:.buttonlessService),
        DFUUuid(withUUID: CBUUID(string: commandCharactUUID), forType:.buttonlessWithoutBondSharing)
    ]

    // ... and ...

        initiator = DFUServiceInitiator().with(firmware: selectedFirmware!)
        initiator!.logger = self
        initiator!.delegate = self
        initiator!.progressDelegate = self
        initiator!.peripheralSelector = self
        initiator!.uuidHelper = DFUUuidHelper(customUuids: dfu_uuid)
        initiator!.alternativeAdvertisingNameEnabled = false
        initiator!.enableUnsafeExperimentalButtonlessServiceInSecureDfu = true // <<<< NEW
```

Again, thanks a lot for you help 👍

connyhald, 2021-02-18:
It's a bit odd, that we need `enableUnsafeExperimentalButtonlessServiceInSecureDfu()` because on Android there is this option as well, but there we do not need it.

Here is our Android code which is working perfectly well.

```kotlin
        val starter = DfuServiceInitiator(mAddress)
        starter.setDeviceName(mName)
        starter.setCustomUuidsForButtonlessDfuWithoutBondSharing(ourServiceUUID, commandCharactUUID)
        starter.setKeepBond(false)
        starter.setNumberOfRetries(3)
        starter.setZip(R.raw.firmware_release)
```

philips77, 2021-02-18:
Yes.. as far as I know the UUID mappings were added in a PR. I'm glad that it works for you!

closed 2021-02-18 by philips77