nordicsemi/IOS-DFU-Library #406: Issue while discovering services

silicusaditi, 2021-03-02:
The services do not get discovered and the device gets disconnected after 30 seconds. However, in the nRF application the advertised services are shown.

philips77, 2021-04-27:
> We used NRF application to check our device. 

What is NRF application? I guess you are talking about nRF Connect for iOS. Why creating the issue in DFU library then?

> The application connects to the device but automatically disconnects after 30 seconds.

This may be related to unsupported connection interval, when the device does not agree to one chosen by iOS. Or some other reason, like intentional disconnection.

> During this timeframe it displays the advertised services.

Are you talking about GATT services aka attribute table (services, characteristics, descriptors), not advertising data? A device advertises usually if it's not connected, then stops advertising when connected (in some use cases it continue to advertise). Advertised services would be the services the device has advertised with using Service UUID field in the advertising packet. They don't have to match the attribute table, so you may advertise with different services, that a device in fact has in its table. Not smart, but possible, as they are independent.

> Here it does not display any services.

So you cannot to service discovery? The "Advertising services" you're talking about, visible in nRF Connect for iOS, are taken from advertising data using this callback: https://developer.apple.com/documentation/corebluetooth/cbcentralmanagerdelegate/1518937-centralmanager

> Can you please explain if there is any difference between code and application.

What code? What application?

philips77, 2021-07-30:
Closing the issue due to inactivity.

closed 2021-07-30 by philips77