nordicsemi/IOS-DFU-Library #407: Customize MTU

Yuuuuu, 2021-04-20:
The [packetSize](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/iOSDFULibrary/Classes/Implementation/SecureDFU/Characteristics/SecureDFUPacket.swift#L61) is automatically set to use the `maximumWriteValueLength`. Is it possible to set the MTU to a lower value? similar to this [method](https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/DfuServiceInitiator.java#L372) on Android.

philips77, 2021-04-20:
I can add it in the next release of you need it. May I ask why do you need it?

Yuuuuu, 2021-04-27:
I'm not sure the exact root cause, think it is a similar issue described [here](https://github.com/NordicSemiconductor/Android-DFU-Library/issues/111).

philips77, 2021-04-29:
As suggested [there](https://github.com/NordicSemiconductor/Android-DFU-Library/issues/111#issuecomment-447883021), it may be enough to set PRN. On iOS you need to set this:
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/d4a1da27fbf195b325cea3036235a4939a8f4efa/iOSDFULibrary/Classes/Implementation/DFUServiceInitiator.swift#L115
which by default is set to 12. In theory, since iOS 11 this can be set to 0 (disabled):
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/d4a1da27fbf195b325cea3036235a4939a8f4efa/Example/iOSDFULibrary/View%20Controllers/DFUViewController.swift#L233-L235
but as you can see, some implementations need it.

I'll release 4.10 without requested feature, but if you find it necessary and test it, we'll think about it.

Yuuuuu, 2021-04-29:
I tried setting a different value for the `packetReceiptNotificationParameter` but still experienced the same issue. 

Do you think it is still valuable to have the ability to set a different MTU independent of the `packetReceiptNotificationParameter`?

philips77, 2021-04-30:
Well.. you may limit the MTU on the bootloader side. All bootloader capabilities should be calculated and tailored so it is possible to do DFU with maximum values without problems. If the flash requires some time to be written to, perhaps the PRN notifications should be used to sync transmitter and receiver. But if even a single packet is too long, or comes too fast with high MTU and DLE enabled, perhaps MTU could be lowered, of DLE disabled. You know what I mean. It should be the device responsibility to support all parameters it says it supports, not phone's to find out lower.