nordicsemi/IOS-DFU-Library #417: Version 4.10.0 of iOSDFULibrary doesn't compile

pmurphyjam, 2021-04-30:
Latest version 4.10.0 doesn't compile in Swift 5
Instantiating it as a Podfile, and it won't compile in Xcode 12.4

**DFU Bootloader version (please complete the following information):**
 - SDK version: [e.g. SDK 16]
 - Bonding used: [yes/no]
 - Library version: [e.g. 4.6.1]

**Device information (please complete the following information):**
 - Device: [e.g. iPhone 11]
 - OS: [e.g. iOS 13.1]

**Describe the bug**
A clear and concise description of what the bug is.

**Logs**
Add any other context or logs describing the problem here.

philips77, 2021-04-30:
It works for me. What errors do you have? Please, provide any logs or screenshots.

philips77, 2021-04-30:
Perhaps you had an issue with not specified architectures? I just released 4.10.1 which should have this fixed. Please try the new version.

pmurphyjam, 2021-04-30:
Installed latest version 4.10.1 still has compile problems.
[image]

pmurphyjam, 2021-05-01:
After reverting to 4.9.0 this is what I see:
[image]

philips77, 2021-05-01:
Ok, I'll fix it in Monday. Can you update to xcode 12.5? I don't have this issue there.

ToHiren, 2021-05-03:
older Version of Xcode create problem below 12

philips77, 2021-05-03:
> older Version of Xcode create problem below 12

Is it the same issue?

philips77, 2021-05-03:
Fixed in 4.10.2. Also, this time I tested it in Xcode 12.4 and seems to be working.

closed 2021-05-03 by philips77