nordicsemi/IOS-DFU-Library #427: How to solve the problem described in #368

dagyu, 2021-06-01:
**Question**
I experienced the problem described in #368 and, even if I read a lot about it, I can't resolve it.

To solve the problem you gave two possible solutions but I want to ask you further resources to implement it, in particular:

- `application not to share the peer data with bootloader`. How I can tell the application to do that? 
- `modifying either the bootloader not to use the direct advertising`. You said that just a `#define` must be changed but how I can find which one is? https://github.com/NordicSemiconductor/Android-DFU-Library/issues/234#issuecomment-693658657

**DFU Bootloader version:**
 - SDK version: nRF5 SDK v11.0.0
 - Bonding used: no
 - Library version: 4.10.3

philips77, 2021-06-01:
Have a look at: https://github.com/NordicSemiconductor/Android-DFU-Library/issues/262#issuecomment-665493850

dagyu, 2021-06-01:
Thank you for the answer!

dagyu, 2021-06-01:
I did the changes but I have still an error, a different one. 

(In sake of completeness I wrote also this code as suggested `dfuInitiator.forceScanningForNewAddressInLegacyDfu = true`)

This is the log:

```
State changed to: Connecting
D: [Callback] Central Manager did update state to: Powered ON
V: Connecting to CLIU_PRO...
D: centralManager.connect(peripheral, options: nil)
D: [Callback] Central Manager did connect peripheral
I: Connected to CLIU_PRO
V: Discovering services...
D: peripheral.discoverServices(nil)
I: Services discovered
V: Starting Legacy DFU...
I: Connected to CLIU_PRO
I: Services discovered
V: Legacy DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
I: DFU characteristics discovered
State changed to: Starting
V: Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
V: Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
A: DFU Control Point notifications enabled
W: Application with buttonless update found
State changed to: Enabling DFU Mode
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected by the remote device
V: Connecting to CLIU_PRO...
D: centralManager.connect(peripheral, options: nil)
D: [Callback] Central Manager did connect peripheral
I: Connected to CLIU_PRO
V: Discovering services...
D: peripheral.discoverServices([00001530-1212-EFDE-1523-785FEABCD123])
I: Services discovered
V: Legacy DFU Service found
V: Discovering characteristics in DFU Service...
D: peripheral.discoverCharacteristics(nil, for: 00001530-1212-EFDE-1523-785FEABCD123)
I: DFU characteristics discovered
V: Reading DFU Version number...
D: peripheral.readValue(00001534-1212-EFDE-1523-785FEABCD123)
I: Read Response received from 00001534-1212-EFDE-1523-785FEABCD123, value (0x): 0800
A: Version number read: 0.8
State changed to: Starting
V: Enabling notifications for 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.setNotifyValue(true, for: 00001531-1212-EFDE-1523-785FEABCD123)
V: Notifications enabled for 00001531-1212-EFDE-1523-785FEABCD123
A: DFU Control Point notifications enabled
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x0104, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
V: Writing image sizes (0b, 0b, 77716b) to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x0000000000000000942f0100, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
A: Start DFU (Op Code = 1, Upload Mode = 4) request sent
I: Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100101
A: Response (Op Code = 1, Status = 1) received
A: Writing Initialize DFU Parameters...
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x0200, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
V: Writing to characteristic 00001532-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x5200ffffffffffff0100a500632d, for: 00001532-1212-EFDE-1523-785FEABCD123, type: .withoutResponse)
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x0201, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
I: Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100201
A: Response (Op Code = 2, Status = 1) received
A: Initialize DFU Parameters completed
State changed to: Uploading
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x080000, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
A: Packet Receipt Notif Req (Op Code = 8, Value = 0) request sent
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x03, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
I: Data written to 00001531-1212-EFDE-1523-785FEABCD123
A: Uploading firmware...
V: Sending firmware to DFU Packet characteristic...
Step progress 0.0
Step progress 0.01
I: Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100306
A: Response (Op Code = 3, Status = 6) received
E: Error 6: Operation failed
V: Writing to characteristic 00001531-1212-EFDE-1523-785FEABCD123...
D: peripheral.writeValue(0x06, for: 00001531-1212-EFDE-1523-785FEABCD123, type: .withResponse)
I: Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100306
A: Response (Op Code = 3, Status = 6) received
E: Error 6: Operation failed
I: Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100306
A: Response (Op Code = 3, Status = 6) received
E: Error 6: Operation failed
I: Notification received from 00001531-1212-EFDE-1523-785FEABCD123, value (0x): 100306
A: Response (Op Code = 3, Status = 6) received
E: Error 6: Operation failed
D: [Callback] Central Manager did disconnect peripheral
I: Disconnected by the remote device
Error 6: Operation failed
```

philips77, 2021-06-01:
The device seems not to be able to consume all the bytes sent with proper speed. I suggest setting Packet Receipt Notifications (PRNs): 
https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/5e41c2155643ced9656b7c1c598f4ba52539e483/iOSDFULibrary/Classes/Implementation/DFUServiceInitiator.swift#L98-L115
If you're disabling them (setting to 0), try not to. If you're using the default value, try setting to 6 or even to 1. This will make upload very slow, but at least should work.

dagyu, 2021-06-01:
Thank you! Now it works.

closed 2021-06-01 by dagyu