nordicsemi/IOS-DFU-Library #428: Pod not building on Xcode 13 Beta & iOS 15 Beta

juliansteenbakker, 2021-06-10:
**Information**
I tried building this pod for a flutter plugin with the Xcode 13 Beta, however it gives a lot of errors. Seems to be something with unwrapping the service from the characteristic object. See the logs for more information.

**DFU Bootloader version:**
 - Library version: 4.10.3

**Device information:**
 - Device: iPad Air 2
 - OS: iOS 15 Beta

**Logs**
(Image because full stack trace is too long for Github)
[image]

philips77, 2021-06-10:
Thank you for creating the issue. I really like Apple for caring about backwards compatibility.

evnik, 2021-06-11:
This seems to be because of this kind of [API changes](https://developer.apple.com/documentation/corebluetooth/cbservice/1434334-peripheral?changes=latest_minor)

juliansteenbakker, 2021-06-11:
At least its not like Android where everything gets deprecated after a new api release.. 😅

dinesharjani, 2021-06-24:
Fixed: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/pull/433

closed 2021-06-24 by dinesharjani