nordicsemi/IOS-DFU-Library #451: Concurrent DFU - Delegate is GC'd

zmoshansky, 2021-08-06:
**Describe the bug**
It appears that when attempting to perform multiple DFU operations concurrently, that my delegate's get garbage collected and stop reporting progress. I have verified that DFU completes by passing in a logger delegate. For some reason, when performing multiple DFU's, the delegate appears to be GC'd and no longer reports progress. However, If I attempt to hold a reference to the delegate until DFU completes, the app crashes as soon as DFU starts.

**Logs**

Initiating class
```
import iOSDFULibrary
@objc(DfuUpdate) class DfuUpdate : CDVPlugin, LoggerDelegate  {
    // Cache firmwares with the same URL to avoid disk writing issues
    var dfus = [String: DfuExecute]();
    // let diskQueue = DispatchQueue(label: "diskQueue")

    @objc(updateFirmware:)
    func updateFirmware(command: CDVInvokedUrlCommand) {
        commandDelegate!.run {
            let dfuCallbackId = command.callbackId

            let options = command.argument(at: 0) as? NSDictionary;
            if(options == nil) {
                self.commandDelegate!.send(
                    CDVPluginResult(
                        status: CDVCommandStatus_ERROR,
                        messageAs: "The first Argument must be the Configuration"
                    ),
                    callbackId: dfuCallbackId
                )
                return;
            }

            let deviceId = options?.value(forKey: "deviceId") as? String
            let fileURL = options?.value(forKey: "fileUrl") as? String;
            let packetReceiptNotificationsValue = options?.value(forKey: "packetReceiptNotificationsValue") as? NSInteger ?? 10;

            // Check Device ID is valid
            if(deviceId == nil) {
                self.commandDelegate!.send(
                    CDVPluginResult(
                        status: CDVCommandStatus_ERROR,
                        messageAs: "Device id is required"
                    ),
                    callbackId: dfuCallbackId
                )
                return;
            }
            if deviceId!.count < 1 {
                self.commandDelegate!.send(
                    CDVPluginResult(
                        status: CDVCommandStatus_ERROR,
                        messageAs: "Device ID is required"
                    ),
                    callbackId: dfuCallbackId
                )
                return
            }
            let deviceUUID = UUID.init(uuidString: deviceId!) ?? nil
            if (deviceUUID == nil) {
                self.commandDelegate!.send(
                    CDVPluginResult(
                        status: CDVCommandStatus_ERROR,
                        messageAs: "Address " + deviceId! + " is not a valid UUID"
                    ),
                    callbackId: dfuCallbackId
                )
                return
            }
            print(deviceUUID, dfuCallbackId)

            // Check file is valid
            if(fileURL == nil) {
[... 156 lines cut ...]
                "percent": progress,
                "speed": currentSpeedBytesPerSecond,
                "avgSpeed": avgSpeedBytesPerSecond,
                "currentPart": part,
                "partsTotal": totalParts,
            ]
            ] as [String : Any]

        print(dfuCallbackId, message)
        let pluginResult = CDVPluginResult(
            status: CDVCommandStatus_OK,
            messageAs: message
        )
        pluginResult?.setKeepCallbackAs(true)
        self.commandDelegate!.send(pluginResult, callbackId: dfuCallbackId)
    }

    func clearHandlers() {
        self.dfuCallbackId = nil
        self.logger = nil
        self.commandDelegate = nil
    }

}
```

philips77, 2021-08-07:
Is the crash coming from the library, or Cordova?
Which library version are you using?
Does is happen 100% times, or occasionally?
There's no GC in swift, but ARC. Delegates are kept weak, so you need to keep strong references, it they'll be released.

zmoshansky, 2021-09-14:
It turns out it was related to Cordova. Now I have multiple concurrent DFU's working. Thanks!

closed 2021-09-14 by zmoshansky