nordicsemi/IOS-DFU-Library #452: New CentralManager instance never gets PoweredOn

Crudough, 2021-09-07:
**Information**
I'm attempting to run a DFU from my iOS application to a Nordic Dev Board in the BL with the Secure DFU Service. After stepping through I've found that the `peripheral.start()` method always returns due to the CentralManager not being .poweredOn. I'm a bit lost as to how this is possible as I have an initial CentralManager up and running on the application scanning, connecting to, and interrogating the characteristics of the BL.

**DFU Bootloader version (please complete the following information):**
 - N/A, custom boot loader but running the DFU service.

**Device information (please complete the following information):**
 - Device: iPhone 11 Pro Max
 - OS: iOS 14.7
 - 
**Your question**
Why is the new Central Manager needed? Can I ask the library to use my manager? (apologies if this is redundant, I've tried reading through the code but haven't found anything yet)

philips77, 2021-09-08:
Hi,
In some older versions the library was stealing `CBCentralManager` instance given in the init in initializer. The problem with that was, that it was reassigning the delegates to the manager and peripheral and was not resetting them back when done. Now, the a manager is created and used in the lib, with own delegates and the peripheral or the central used to obtain it are not touched.

Crudough, 2021-09-08:
@philips77 Thank you for the explanation! I understand the reasoning now. I initially thought that this was causing the DFU process to not start however I've realized that the DFU completes successfully but the lack of feedback was leading me to believe that it had failed. Please see Issue #453 which describes the issue the I am experiencing.