nordicsemi/IOS-DFU-Library #453: Logs only present while stepping through DFU

Crudough, 2021-09-08:
**DFU Bootloader version (please complete the following information):**
  - Library version: 4.11.0

**Device information (please complete the following information):**
 - Device: iPhone 11 Pro Max
 - OS: iOS 14.7

**Your question**
I only get logs in my Xcode console if I step through the program which, correctly, results in a connection timeout for the DFU process. I can confirm that a DFU indeed takes place and am using the CPU usage as the indicator that the process is done. I only receive one "Connecting" log initially when I prompt the DFU but nothing after that. I've included some code snippets below and the "Connecting" log.

**Logs**
*Code Snippet*

```class DFUManager: DFUServiceDelegate, DFUProgressDelegate, LoggerDelegate {
    
    /* Class Properties */
    private var dfuPeripheral   : CBPeripheral!
    private var peripheralName  : String?
    private var dfuController   : DFUServiceController!
    private var dfuInitiator    : DFUServiceInitiator!
    private var dfuFirmware     : DFUFirmware!
    
    private var dfuProgress: Int = 0
    
    init(peripheral: Peripheral, dfuUrl: URL) {
        self.dfuPeripheral = peripheral.manager.connectedPeripheral
        self.peripheralName = peripheral.name
        guard
            let firmware = DFUFirmware(urlToZipFile: dfuUrl) else {
                print("Cannot get DFU firmware for url: \(dfuUrl).")
                return
            }
        self.dfuFirmware  = firmware 
        self.dfuInitiator = DFUServiceInitiator(queue: DispatchQueue(label: "DFU Queue").with(firmware: self.dfuFirmware)
    }
    
    /* DFUServiceDelegate */
    func dfuStateDidChange(to state: DFUState) {
        switch state {
        case .completed, .disconnecting:
            /* Throw completion flag, forget the controller */
            print("I have completed the DFU!")
            dfuController = nil
        case .aborted:
            /* Throw abort status flag, forget the controller */
            dfuController = nil
        case .connecting:
            /* Should not happen, the peripheral should be connected. */
           // >>>>>> This is the only log from any delegate methods that every gets called.
            print("Connecting")
        default:
            /* Nothing had happened, keep the option to stop/abort open */
            print("Default")
        }
    }
    
    func dfuError(_ error: DFUError, didOccurWithMessage message: String) {
        print("Error \(error.rawValue): \(message)")
                
        // Forget the controller when DFU finished with an error
        dfuController = nil
    }
    
    /* DFUProgressDelegate */
    func dfuProgressDidChange(for part: Int, outOf totalParts: Int, to progress: Int, currentSpeedBytesPerSecond: Double, avgSpeedBytesPerSecond: Double) {
        dfuProgress = progress
        
        /* update views here or more state variables that will affect front end views */
        print("DFU Progress: \(progress).")
        print("DFU Part \(part) out of \(totalParts).")
        print("Current Speed: \(currentSpeedBytesPerSecond)")
        print("Average Speed: \(avgSpeedBytesPerSecond).")
    }
    
    func startDFUProcess() {
        
        guard let dfuPeripheral = self.dfuPeripheral else {
            print("No DFU peripheral was set")
            return
        }
        
        self.dfuInitiator.delegate = self
        self.dfuInitiator.progressDelegate = self
        self.dfuInitiator.logger = self
        self.dfuInitiator.enableUnsafeExperimentalButtonlessServiceInSecureDfu = true
        guard let controller = self.dfuInitiator.start(target: dfuPeripheral) else {
            print("Could not start controller!")
            return
        }
        print("\nI am starting the DFU Process")
        self.dfuController = controller
    }
    
    func logWith(_ level: LogLevel, message: String) {
        print("\(level.name()): \(message)")
    }
}
```

*Simple Logs*
```
I am starting the DFU Process
Connecting
```

philips77, 2021-09-08:
Hello,
Regarding why the state goes to `.connecting`, each client (`CBCentralManager`) may connect to a device separately and they are not aware of each other. The OS system manages the physical connection, but informs clients as if they were the only client connected, to some degree. The state changes to `connecting` even before anything starts: 
https://github.com/NordicSemiconductor/IOS-DFU-Library/blob/8f935c6516969a874d8c2c01f4fdc4188c3d9910/iOSDFULibrary/Classes/Implementation/DFUSelector/DFUServiceSelector.swift#L69-L74

Could you debug why you don't get logger events, and where the process actually stops? This method should be called
https://github.com/NordicSemiconductor/IOS-DFU-Library/blob/8f935c6516969a874d8c2c01f4fdc4188c3d9910/iOSDFULibrary/Classes/Implementation/GenericDFU/DFUPeripheral.swift#L158-L175 and the state should be `.connected`, of the `connect()` method should be called.
As you don't get `Connected to NAME` log, I guess it starts to connect, but than you should get this log:
https://github.com/NordicSemiconductor/IOS-DFU-Library/blob/8f935c6516969a874d8c2c01f4fdc4188c3d9910/iOSDFULibrary/Classes/Implementation/GenericDFU/DFUPeripheral.swift#L475-L480

Crudough, 2021-09-08:
@philips77 I've stepped through and it seems that it's not picking up the logger peripheral in LoggerHelper.swift.

https://github.com/NordicSemiconductor/IOS-DFU-Library/blob/d4a1da27fbf195b325cea3036235a4939a8f4efa/iOSDFULibrary/Classes/Utilities/Logging/LoggerHelper.swift#L76-L82

The logger.X calls happen but the Logger Helper cannot resolve the logger - the `if let logger = logger` does not execute. As far as I'm aware I believe I am connecting the delegate correctly.

Crudough, 2021-09-08:
As additional information. Right before the code snippet for the DFUServiceSelector.swift file that you've included, in the init, I can confirm that it is indeed my logger where it is referenced as `initiator.logger`.

https://github.com/NordicSemiconductor/IOS-DFU-Library/blob/8f935c6516969a874d8c2c01f4fdc4188c3d9910/iOSDFULibrary/Classes/Implementation/DFUSelector/DFUServiceSelector.swift#L60-L67

Crudough, 2021-09-08:
I've updated the issue above to include the full logs while the code is stepped through.

Crudough, 2021-09-08:
To leave a trail and hints for anyone else that may be stuck with this issue. I've found that this seems to be a timing issue. Initially I was instantiating a `DFUManager` from my code snippet in my original description then immediately starting the DFU process. I've since changed the architecture to instantiate the DFUManager in an `.onAppear` closure and waiting until a second action (button) to start the DFUProcess itself. It appears that this is logging correctly now! 

@philips77 I will leave this open for a day or two if you are able to clarify why this might be the case. Thank you for all the help!

closed 2021-09-10 by Crudough