nordicsemi/IOS-DFU-Library #454: Xcode13 can't not

IrisMo, 2021-09-18:
**Information**
It can be used normally before. When I upgraded to big sur 11.6, xcode13, the SDK will report the following error when compiling, and the demo of the mac version can be reproduced. Will you be able to adapt later? Or what can I do to avoid this situation? Hope to get your reply, thank you!

[image]

dinesharjani, 2021-09-20:
Thanks for bringing this up @IrisMo - we just merged a fix to the develop branch: https://github.com/NordicSemiconductor/IOS-DFU-Library/pull/455

Try it out when you can.

closed 2021-09-21 by dinesharjani

stami-fb, 2021-09-22:
@philips77 @dinesharjani can we get a Cocoapods release from this Xcode 13 compatible version, please? 🙏 

I'm using react-native-nordic-dfu which has this repo as it's Cocoapods spec dependency, and those cannot be pointed to a git branch (because of how Cocoapods works). ☹️  So this is preventing anyone from using that on Xcode 13 right now...

stami-fb, 2021-09-22:
For those ending up here with the same problem, I am able to temporarily work around this issue by removing/commenting out the spec dependency from react-native-nordic-dfu podspec
```
# s.dependency 'iOSDFULibrary', '~> 4.7.1'
```

and adding direct dependency to my main app's Podfile to this develop branch:
```
pod 'iOSDFULibrary', :git => 'https://github.com/NordicSemiconductor/IOS-DFU-Library.git', :branch => 'develop'
```

philips77, 2021-09-22:
I'll do this tomorrow.

spaceMan00, 2021-10-05:
@philips77 has this fix been released for cocoapods? I still cannot build on xcode 13 for iOS. Getting similar errors
[image]

Matheuschn, 2021-10-05:
> @philips77 @dinesharjani has this fix been released for cocoapods? I still cannot build on xcode 13 for iOS. Getting similar errors [image]

I am also having the same problem. The way i fixed it, at least until the fixes get to Cocoapods, is by replacing the library's folder (Pods/IOSDFULibrary) with a clone from the develop branch (make sure the names match). Then, run 'pod install' and it should work for now.

@philips77 @dinesharjani, if possible, can you update the main branch with the fixes? Thanks!

philips77, 2021-10-05:
I'm on it!

philips77, 2021-10-05:
Fix released in 4.11.1.