nordicsemi/IOS-DFU-Library #456: iOS15 compile error

liyonghui16, 2021-09-22:
CBService 's property is changed to optional
`weak open var peripheral: CBPeripheral? { get }`

emoonadev, 2021-09-22:
Same problem in Xcode 13

emoonadev, 2021-09-22:
Merged fix to develop for the [issue](https://github.com/NordicSemiconductor/IOS-DFU-Library/issues/454#issuecomment-922871516)

philips77, 2021-09-22:
Are you developing for iOS or Mac?

philips77, 2021-09-22:
This change has been addressed in 4.10.4, as far as i remember. But seems like Apple made it more interesting and now the API diverged between iOS and Mac. We #455 tries to address that issue no matter the target.

liyonghui16, 2021-09-23:
Sorry, I'm using an older version。😂 @philips77

closed 2021-09-23 by liyonghui16

rodmaz, 2021-09-30:
We are using version 4.11.0 (Swift Package) and still getting these issues when the app is compiled in release mode/archive.