nordicsemi/IOS-DFU-Library #458: Double firmware and Buttonless DFU

fiveagle, 2021-09-23:
Hi,
On the project I am working on we have the necessity to upload 2 firmware to the device. One is a classic firmware update and the other is a STM update for the not bluetooth board. For what I could understand the proposal from the firmware provider is to execute 2 dfu process, one after the other for each firmware, but the second one is required to be buttonless and so easier for the final user.
So my questions are:
How should I initiate the buttonless dfu process?
Is this double dfu strategy the right approach to the problem?

Thanks.

philips77, 2021-09-24:
Hello @fiveagle,

As you're asking here, I believe your software is based on the [nRF5 SDK](https://www.nordicsemi.com/Products/Development-software/nRF5-SDK), not [nRF Connect SDK](https://www.nordicsemi.com/Products/Development-software/nRF-Connect-SDK). Just FYI, for the latter [this](https://github.com/NordicSemiconductor/Android-nRF-Connect-Device-Manager) library should be used instead.

Also, I'm guessing your SDK is fairly new, so we're talking about Secure DFU (nRF5 SDK 12+), not Legacy DFU (nRF5 SDK 4.3-11).

Secure DFU has Buttlonless service explained [here](https://infocenter.nordicsemi.com/topic/sdk_nrf5_v17.1.0/ble_sdk_app_buttonless_dfu.html?cp=8_1_4_4_1). This library (and DFU Library for Android) support it in both configurations: for bonded and not bonded devices. 

> Note, that iOS has no clue whether the device is bonded, or not, so it's up to the firmware on the peripheral side to ensure bonding.

This library will connect to your device (the client (local `CBCentralManager`) will connect, the device doesn't need to be disconnected; it may be connected with another central manager instance), discover services  and, if buttonless service is found, trigger jump to bootloader mode automatically. 

> Note: you may also customize service UUIDs if not using the standard ones.

The double DFU seems the right approach, yes. But, I believe, your firmware (bootloader) needs somehow distinguish those updates. Which fw should be applied to nRF5 chip, and which to the other one. This can be done by e.g. by modifying the structure (*dfu-cc.proto*) of the [init packet](https://infocenter.nordicsemi.com/topic/sdk_nrf5_v17.1.0/lib_bootloader_dfu_validation.html?cp=8_1_3_5_1_1_0#lib_bootloader_dfu_init), using different parameters of the existing properties (e.g hardware version), or some other way. The point is, that both updates may look the same from the mobile app point of view and no changes in the protocol is required, so you may, as you said, perform two updated. I understand, that after the first update the device would reboot to app mode, with Buttonless service present, which would then allow another update. In your app you need to check in which state are you, as the DFU may be interrupted in the middle, so you need to know which ZIP file provide for which update. E.g. your device disconnects during the first DFU, so it should resume with the same ZIP including the nRF5 firmware, not for the other chip.

philips77, 2022-06-30:
Closing due to inactivity.