nordicsemi/IOS-DFU-Library #459: DFU works on Nordic Development Kit but not on Custom Board (BL652)

peterg0313210, 2021-09-25:
**DFU Bootloader version (please complete the following information):**
 - SDK version: SDK 17
 - Bonding used: I am assuming no as in my current config NRF_DFU_BLE_BUTTONLESS_SUPPORTS_BONDS = 0 in sdk_config.h
 - Library version: 7.0.1

**Device information (please complete the following information):**
 - Device: Android, IPad
 - OS: Latest OS on both devices

**Your question**
Hello, I added a DFU service to some code I wrote for an NRF52382 based BL652 board using segger embedded studio. I have been testing my firmware on a nordic development board before implementing code on final board. I followed the pdf tutorial, Add Nordic Semiconductor DFU to SDK example, and followed all the steps almost verbatim from section 5 down, talking about implementing buttonless bootloader, I ran the code on my development board, and it worked like a charm on the development board. 

I figured, oh well this must now be able to work on the custom board as well. I tested there and I always get the DFU disconnecting while trying to upload firmware. I made sure my dfu package was compatible and everything, my dfu package zip uploaded to my development board with the same exact firmware running and it works perfectly every time on the nordic board. The issue seems to be that when the buttonless dfu service sends 0x01 and receives 0x200101 right at that moment the DFU disconnects. I have the log shown below from my iPad for reference. I did disable alternative advertising mode as is often recommended. To me it seems strange that on the nrf52832 dev board the dfu service works with pretty much 100 percent reliability and on custom board which uses bl652, nrf52382 based, the dfu service shows up of course but DFU disconnects immediately when it is trying to enter bootloader mode. Both times I am using the same dfu_package.zip so I highly doubt there is an issue with the zip file. I am just trying to see where the issue could be and see if you know of any places to start looking in my firmware for debugging. Perhaps I am missing something regarding compatibility from bl652 to Nordic development board. 

Any help would be greatly appreciated. Thank you 

**Logs**
.
[image]

[Add-NordicSemiconductorDFU-to-SDKexample_AppNote_UBX-19050198 (1).pdf](https://github.com/NordicSemiconductor/IOS-DFU-Library/files/7230293/Add-NordicSemiconductorDFU-to-SDKexample_AppNote_UBX-19050198.1.pdf)

peterg0313210, 2021-09-25:
Just as an additional comment I tested again by actually, purposefully, writing 0x01 into buttonless dfu service and it returns 0x200 101 but then it disconnects immediately and when I reconnect and check the log it says error 8(0x8): GATT CONN TIMEOUT

peterg0313210, 2021-09-27:
Of course I am not completely sure this is the problem but it is a worthwhile thing to check out.

philips77, 2022-06-30:
Hi,
Sorry for coming so late. I was busy with other projects.

The command you're sending should restart the device in DFU mode, so it should switch to bootloader and start advertising (usually with MAC address +1, so visible as a different device).
In your case there may be different reasons for the observed behavior:
1. You didn't flash the DFU bootloader on your custom board. Ensure it is flashed and you may trigger it.
2. The bootloader advertises with data that are not expected by the library by default, try implementing custom `DFUPeripharalSelector` if this is the case. The default one is scanning for a device with 0xFE59 Service UUID.
3. Your buttonless setup doesn't work, so you may need to fix it. Try switching to the bootloader mode with a button, if your setup supports it.

I'm closing the issue due to inactivity, hoping you have solved the issue long time ago.

closed 2022-06-30 by philips77