nordicsemi/IOS-DFU-Library #470: logWith --- message = Error 12: Verification failed

zhoumt, 2022-01-14:
Hi, big guy, I'm a new developer! After I download the firmware from the Internet for DFU upgrade, there is an error of verification failure. May I ask what is the cause? Looking forward to your reply, much appreciated!

The following is my file path：
zipPath = file:///var/mobile/Containers/Data/Application/CBEDDED3-8625-4564-9273-E4916F461459/Documents/pump.zip

dinesharjani, 2022-01-14:
Sp @zhoumt 'verification' I don't think is because the file can't be opened, it's more like in the DFU process. Can you share with us the file causing the issue?

zhoumt, 2022-01-14:
ok, please receive the file,thanks very much!
[pump.zip](https://github.com/NordicSemiconductor/IOS-DFU-Library/files/7869963/pump.zip)
.

philips77, 2022-01-14:
Your zip file seems invalid. 
> Bad CRC-32 for file 'ble_insulin_pump_p.bin'

Use https://github.com/NordicSemiconductor/pc-nrfutil to create a valid one from your hex file. Provide correct validation parameters like required SD, app version, HW version, etc.

closed 2022-01-14 by philips77