nordicsemi/IOS-DFU-Library #472: Init Packet Disection on iOS

Crudough, 2022-03-01:
**DFU Bootloader version (please complete the following information):**
 - SDK version: N/A
 - Bonding used: N/A
 - Library version: 4.11.1

**Device information (please complete the following information):**
 - Device: N/A
 - OS: N/A

**Your question**
Waiting on [PR#396](https://github.com/NordicSemiconductor/IOS-DFU-Library/pull/396) to expose the initPacket as public through the `DFUFirmware` class. In the meantime, I have forked the repo, made the change locally and continued my testing and development. I am unsure as to how best to dissect the information stored in the `initPacket` with my current interest being the `type` field. 

I have been utilizing the protobuf message definitions listed in Nordic's nrfutil repo [here.](https://github.com/NordicSemiconductor/pc-nrfutil/blob/master/nordicsemi/dfu/dfu-cc.proto) The current workflow I'm attempting is as follows, select my DFU Package -> create my `DFUFirmware` object in code through the `DFUManager`-> if successful then deserialize the initPacket into the `InitCommand` to find the packet type (**) -> continue with the rest of the DFU as normal.

(**) This deserialization into the protobuf structure is successful however I cannot seem to extract the type out of the `initPacket` which I have confirmed as `.externalApplication` through `nrfutil pkg display` command.

philips77, 2022-03-02:
Why do you want to get the *type*, if I may ask?

Also, why do you need #396 exactly? For sending the firmware to Flutter? Serializing to protobuf?

Crudough, 2022-03-02:
Hi @philips77,

Thank you for responding here!
I need a way to programmatically detect the application type from the initPacket data to direct internal behaviour and sequencing. We use this in other ways but not through the iOS DFU Manager (ex. by using `nrfutil pkg display` as mentioned above).

As far as I'm aware #396 is the only way of accessing the initPacket from an incoming DFU package. After accessing the initPackage I am hoping to deserialize the data to protobuf using `InitComand(serializedData: stream.initPacket` from the Swift code generated from `dfu_cc.proto`. If there is an alternate way to access the type that does not drive a change please do let me know. Forking and manipulating the class to test has proven problematic with the Swift Package Manager.

philips77, 2022-03-02:
So you'll be more interested in getting the fw type than the data, right?

Crudough, 2022-03-07:
Correct, apologies for the slow response!

philips77, 2022-03-08:
I can see what can be done. Thank you for opening the issue.

Crudough, 2022-03-14:
Appreciate that @philips77! Looking forward to hearing back from you!