nordicsemi/IOS-DFU-Library #473: DFU issue in iOS

gugagop, 2022-04-01:
**DFU Bootloader version :**
 - SDK version: SDK 12.3
 - Bonding used: no
 - Library version:  4.11.1

**Device information:**
 - Device: iPhone 7
 - OS: iOS 15.2

**Your question**
We are using the lib to update nrf51 chips using DFU. With android, it is working fine, but with iOS, we are having a issue that we start sending firmware to DFU Packet characteristic but it stays like that in a loop. Is there anything I have to do to DFU work with iOS specifically? Down there we have the generated Logs.

**Logs**
[Callback] Central Manager did update state to: Powered ON
Connecting to KRemote...
centralManager.connect(peripheral, options: nil)
[Callback] Central Manager did connect peripheral
Connected to KRemote
Discovering services...
peripheral.discoverServices(nil)
Services discovered
Starting Secure DFU...
Connected to KRemote
Services discovered
Secure DFU Service found
Discovering characteristics in DFU Service...
peripheral.discoverCharacteristics(nil, for: FE59)
DFU characteristics discovered
Enabling notifications for 8EC90001-F315-4F60-9FB8-838830DAEA50...
peripheral.setNotifyValue(true, for: 8EC90001-F315-4F60-9FB8-838830DAEA50)
Notifications enabled for 8EC90001-F315-4F60-9FB8-838830DAEA50
Secure DFU Control Point notifications enabled
Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0x0601, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600601000100000000000000000000
Command object selected (Max size = 256, Offset = 0, CRC = 00000000) received
Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0x01018d000000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600101
Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0x020000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600201
Packet Receipt Notif disabled (Op Code = 2, Value = 0)
Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0x128a010a4408011240081210331a028701200028, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0x00300038b8ea0742240803122068fe013af799bb, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0xa0a0517446482964183a6a8e9615b8247e4f757e, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0x0605ec41d5480052040801120010001a40152e1c, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0x2055596157c8fd34851eb2bfa8fc6188d35b8ece, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0xe7059d594a0331b8d14c79e8fecf23b01e2eebb4, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0x54c897335bf64e018ef8b07953187106cea7a3f4, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
Writing to characteristic 8EC90002-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0xf5, for: 8EC90002-F315-4F60-9FB8-838830DAEA50, type: .withoutResponse)
Command object sent (CRC = A84F528C)
Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0x03, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 6003018d0000008c524fa8
Checksum (Offset = 141, CRC = A84F528C) received
Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0x04, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600401
Command object executed
Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0x020c00, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600201
Packet Receipt Notif enabled (Op Code = 2, Value = 12)
Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0x0602, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600601001000000000000000000000
Data object selected (Max size = 4096, Offset = 0, CRC = 00000000) received
Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50...
peripheral.writeValue(0x010200100000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600101
Data object 1/32 created
wait(400)
Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50
Uploading firmware...
Sending firmware to DFU Packet characteristic...

philips77, 2022-04-03:
Where exactly the loop is happening? Do you have more logs?
The ones you've sent look correct, but incomplete.

philips77, 2022-04-04:
PRNs are required if you're supporting iOS version before 11.
In iOS 11 two methods have been added for controlling buffer overflow: 
* `canSendWriteWithoutResponse` -> [link](https://developer.apple.com/documentation/corebluetooth/cbperipheral/2891512-cansendwritewithoutresponse)
* `peripheralIsReady(toSendWriteWithoutResponse:)` -> [link](https://developer.apple.com/documentation/corebluetooth/cbperipheraldelegate/2874034-peripheralisready)

gugagop, 2022-04-04:
Thank you so much! The problem was actualy it. I changed the packet receipt notif enabled to 0 and it worked. Now I know that the problem is something related to that in my firmware.

closed 2022-04-04 by gugagop