nordicsemi/IOS-DFU-Library #476: Add CBCentralManager options to DFUServiceInitiator initializer

mateusforgi, 2022-05-17:
**DFU Bootloader version (please complete the following information):**
 - Library version: 4.10.4

**Device information (please complete the following information):**
 - Device: iPhone XR
 - OS: iOS 15.4.1

**Describe the bug**
The DFUServiceInitiator initializer does not offer an option to pass CBCentralManagerOptions (unless we use the deprecated one), so we cannot use CBCentralManagerOptionShowPowerAlertKey set to false resulting in the user getting a modal asking permission for bluetooth connections.

[image]

mateusforgi, 2022-06-17:
@philips77 are you guys planning a new release soon?

philips77, 2022-06-29:
I'll try today.

closed 2022-06-29 by philips77