nordicsemi/IOS-DFU-Library #479: Resume sending InitPacket doesn't work when offset > (initpacketlength - offset)

kamiljanowski-silvair, 2022-05-20:
**DFU Bootloader version (please complete the following information):**
 - SDK version: 14
 - Bonding used: no
 - Library version: 4.11.1

**Device information (please complete the following information):**
 - Device: iPhone 13
 - OS: iOS 15.4.1

**Describe the bug**

Hi!

Steps to reproduce:

- Start DFU of target device
- Breaks DFU on sending InitPacket when offset of sent packages > 1/2 * InitPacketLength
- Try to resume DFU
- DFU failed with log (i.e. for offset = 88 ) "DFU error 0x1F4 - Asset offset (88) < initPacketLength (171) - offset (88) failed"

I assume that this guard is incorrect, it asserts when offset is valid and DFU should be resumed:
[image]

This guard was introduced in https://github.com/NordicSemiconductor/IOS-DFU-Library/commit/976dd11b274b5d639f487a8d88af159d4fc6120d commit.

**Logs**
```
{"message":"DFU peripheral.writeValue(0x0601, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)"}
{"message":"DFU Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50"}
{"message":"DFU Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 60060100040000580000005c8206a3"}
**{"message":"DFU Command object selected (Max size = 1024, Offset = 88, CRC = A306825C) received"}**
**{"message":"DFU Resuming sending Init packet..."}**
{"message":"DFU Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50..."}
{"message":"DFU peripheral.writeValue(0x020000, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)"}
{"message":"DFU Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50"}
{"message":"DFU Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600201"}
{"message":"DFU Packet Receipt Notif disabled (Op Code = 2, Value = 0)"}
{"message":"DFU Writing to characteristic 8EC90001-F315-4F60-9FB8-838830DAEA50..."}
{"message":"DFU peripheral.writeValue(0x0c, for: 8EC90001-F315-4F60-9FB8-838830DAEA50, type: .withResponse)"}
{"message":"DFU Data written to 8EC90001-F315-4F60-9FB8-838830DAEA50"}
{"message":"DFU Notification received from 8EC90001-F315-4F60-9FB8-838830DAEA50, value (0x): 600c03"}
{"message":"DFU Error 3: Invalid parameter"}
{"message":"DFU Disconnecting..."}
{"message":"DFU centralManager.cancelPeripheralConnection(peripheral)"}
{"message":"DFU [Callback] Central Manager did disconnect peripheral"}
{"message":"DFU Disconnected"}
**{"message":"DFU error 0x1F4 - Asset offset (88) < initPacketLength (171) - offset (88) failed"}**
```

philips77, 2022-05-20:
Thank you. I'll look at it next week.

kamiljanowski-silvair, 2022-06-06:
Hi, any updates?

philips77, 2022-06-06:
I'll do a release tomorrow, sorry! I was busy with other tasks.

kamiljanowski-silvair, 2022-06-27:
Hello, I want to ask about status. Are you able to tell me when are you planning to release a fix for this issue?

philips77, 2022-06-28:
I'm planning to create a release tomorrow or the day after. Once again, sorry for the delay.

closed 2022-06-29 by philips77

philips77, 2022-06-30:
Released as 4.12.0. Please, test.

kamiljanowski-silvair, 2022-06-30:
Thanks for info! I will test it today or tomorrow and give you a feedback.

kamiljanowski-silvair, 2022-06-30:
I tested it manually, and it works :) thank you.

philips77, 2022-06-30:
Btw, I think there may be another release today with more improvements.
Expect PRs soon. One may break `DFUFirmware` API. A bit. In a good way ;)