nordicsemi/IOS-DFU-Library #489: Need new version of Swift.

jjs-home, 2022-08-01:
I have been trying today to update our app and I'm getting a problem with the iosDFULibrary as it was built using Swift 5.3.2. Is it possible to get this using 5.5.2? Here is the error message from XCode.
Module compiled with Swift 5.3.2 cannot be imported by the Swift 5.5.2 compiler

philips77, 2022-08-02:
How are you including the library in your project?
1. CocoaPods
2. Swift Package Manager
3. Carthage
4. Other

What Xcode version are you using?

I'll be releasing a new version of the library in coming days/weeks.

jjs-home, 2022-08-02:
Hi Aleksander,

 

                I have tried with CocoaPods. And I’m using XCode 13.2.1.

 

Cheers,

 

John

jjs-home, 2022-08-02:
Hi Aleksander,

 

                I’ve just updated to XCode 13.4.1 and similar problem. Module compiled with Swift 5.3.2.cannont be imported by the Swift 5.6.1 compiler.

 

Cheers,

 

John

mldesanctis-cgi, 2022-08-12:
Same. I had reported this under https://github.com/NordicSemiconductor/IOS-DFU-Library/issues/475.
I am using Xcode 13.4.1 and including the library using Carthage

zhxf2012, 2022-09-01:
Same. 
I am using Xcode 13.4.1 and including the library using Cocoapods.

jjs-home, 2022-09-01:
I managed to get this working by building using Carthage and copying the xcframework files into the XCode project.