nordicsemi/IOS-DFU-Library #490: DFUServiceDelegate probelm

LucaDonetti, 2022-09-30:
**Information**
I try to update a firmware with the library and use the same method of your example app:

self.dfuInitiator = DFUServiceInitiator(queue: DispatchQueue(label: "FirmwareUpdate"))
self.dfuInitiator?.delegate = self
self.dfuInitiator?.progressDelegate = self
self.dfuInitiator?.logger = self
self.dfuInitiator?.dataObjectPreparationDelay = 0.4
self.dfuInitiator?.packetReceiptNotificationParameter = 0

**DFU Bootloader version (please complete the following information):**
 - SDK version: [e.g. SDK 16]
 - Library version: [e.g. 4.13.0]

**Device information (please complete the following information):**
 - Device: [e.g. iPhone 13]
 - OS: [e.g. iOS 15, iOS 16]

**Describe the bug**
The problem is that I set the delegate and the delegateProcess, but they are nil into the DFUServiceSelector.
Into DFUServiceInitiator: self.initiator.delegate is not nil, but into DFUServiceSelector: self.initiator.delegate == nil. For this 
the callback for the state and the progress not work

**Logs**
Add any other context or logs describing the problem here.

philips77, 2023-02-23:
How and why do you get access to `DFUServiceSelector`? The selector is started with the initiator instance so all the fields should be set as well, unless you set them back to *nil* of the `self` from your code was disposed.

philips77, 2023-02-23:
Also, sorry for answering so late. I am busy with lots of projects and have missed your question. Did you manage to solve the issue? Can we close it?