nordicsemi/IOS-DFU-Library #491: carthage build failure

kylewhittington, 2022-10-06:
**DFU Bootloader version (please complete the following information):**
 - SDK version: n/a
 - Bonding used: n/a
 - Library version: 4.13.0

**Device information (please complete the following information):**
 - Device: n/a
 - OS: n/a

**Describe the bug**
Building from carthage is throwing an error but the xcodebuild log shows a success at the end of it. I have tried altering xcode versions (screenshot you'll see is 13.4.1) but the most recent 14 version is also throwing the same error so I don't think that is it. I have also tried running against the previous two versions of the library with the same issue.

**Logs**
[image]

[image]

kylewhittington, 2022-10-19:
@philips77 any suggestions for this? We're really needing this .framework file that it generates and just can't for the life of me figure out why this won't work anymore.

philips77, 2022-10-19:
Perhaps you can have a look at: https://github.com/weichsel/ZIPFoundation#carthage

> After adding ZIPFoundation to the Cartfile, you have to fetch the sources by running:
   carthage update --no-build

baddinosaur-kyle, 2023-01-31:
Thanks @philips77 -- I believe this was related to an Xcode version causing trouble. Not having any issues on recent Xcode 14.1 whilst doing this. Happy for this to be closed!

closed 2023-02-01 by philips77