nordicsemi/IOS-DFU-Library #496: ios 15 does not respond when starting the dfu process？

Jack-QMY, 2022-12-30:
react-native v0.55.4;
"react-native-nordic-dfu v2.0.1;
ios15.2

It can be used normally on the android platform, and there are related progress，But on the Apple phone, there is no response to executing startDfu(),I checked the moudules file RNNordicDfu.m file,retrievePeripheralsWithIdentifiers returns 0 so the DFU process quits with error "Could not find device with deviceAddress"

NSArray<CBPeripheral *> * peripherals = [centralManager retrievePeripheralsWithIdentifiers:@[uuid]];

image
Then what do I do next？

philips77, 2023-01-02:
If you don't have the UUID associated with your`CBPeripheral` you need to scan for it using `scanForPeripherals`

closed 2023-01-02 by philips77

Jack-QMY, 2023-01-04:
@philips77  I try to replace，     but   Unknown type name 'scanForPeripherals'