nordicsemi/IOS-DFU-Library #499: ios appstore failed

Jack-QMY, 2023-01-09:
I put it on the appstore, and the following problem occurs？
Path:                /private/var/containers/Bundle/Application/0DDCD8ED-47FD-48B7-86D0-E4D892075BF4/wristech.app/wristech
Identifier:          com.weiyun.ShareTreatment
Version:             5 (1.2.5)
AppStoreTools:       11E608a
Code Type:           ARM-64 (Native)
Role:                Foreground
Parent Process:      launchd [1]
Coalition:           com.weiyun.ShareTreatment [703]

Date/Time:           2020-10-11 00:46:14.9878 +0800
Launch Time:         2020-10-11 00:46:14.9806 +0800
OS Version:          iPhone OS 14.0.1 (18A393)
Release Type:        User
Baseband Version:    n/a
Report Version:      104

Exception Type:  EXC_CRASH (SIGABRT)
Exception Codes: 0x0000000000000000, 0x0000000000000000
Exception Note:  EXC_CORPSE_NOTIFY
Termination Description: DYLD, dyld: Using shared cache: 9BCD572E-A399-3F6E-802F-E8A6DFCCACA2 | dependent dylib '@rpath/libswiftCore.dylib' not found for '/private/var/containers/Bundle/Application/0DDCD8ED-47FD-48B7-86D0-E4D892075BF4/wristech.app/Frameworks/Zip.framework/Zip', tried but didn't find: '/private/var/containers/Bundle/Application/0DDCD8ED-47FD-48B7-86D0-E4D892075BF4/wristech.app/Frameworks/libswiftCore.dylib' '/private/var/containers/Bundle/Application/0DDCD8ED-47FD-48B7-86D0-E4D892075BF4/wristech.app/Frameworks/Zip.framework/Frameworks/libswiftCore.dylib' '@rpath/libswiftCore.dylib' '/usr/local/lib/libswiftCore.dylib' '/usr/lib/libswiftCore.dylib'
Highlighted by Thread:  0

Backtrace not available

Unknown thread crashed with ARM Thread State (64-bit):
    x0: 0x0000000000000006   x1: 0x0000000000000001   x2: 0x000000016f872000   x3: 0x00000000000000a8
    x4: 0x000000016f871c00   x5: 0x0000000000000000   x6: 0x0000000000000000   x7: 0x0000000000000000
    x8: 0x0000000000000020   x9: 0x0000000000000009  x10: 0x0000000000000001  x11: 0x0000000000000057
   x12: 0x000000000000002e  x13: 0x00000000000000a3  x14: 0x0000000000000099  x15: 0x000000000000003f
   x16: 0x0000000000000209  x17: 0x0000000100ced034  x18: 0x0000000000000000  x19: 0x0000000000000000
   x20: 0x000000016f871c00  x21: 0x00000000000000a8  x22: 0x000000016f872000  x23: 0x0000000000000001
   x24: 0x0000000000000006  x25: 0x000000016f8728c8  x26: 0x0000000100d14778  x27: 0x0000000100d14000
   x28: 0x0000000000000000   fp: 0x000000016f871bd0   lr: 0x0000000100cfa3ac
    sp: 0x000000016f871b90   pc: 0x0000000100cf2bbc cpsr: 0x00000000
   esr: 0x00000000  Address size fault

Binary images description not available

Error Formulating Crash Report:

EOF

[image]

philips77, 2023-01-09:
How about https://stackoverflow.com/a/26949219/2115352?

Jack-QMY, 2023-01-10:
> https://stackoverflow.com/a/26949219/2115352怎么样？
But I also set Build Settings > Always Embed Swift Standard Libraries to YES, and the appstore has no effect
[image]

philips77, 2023-02-23:
I'm closing the issue as it is related to xcode configuration, not the library itself. If you still haven't found the solution I recommend Stack Overflow or other services.

closed 2023-02-23 by philips77