nordicsemi/IOS-DFU-Library #501: NordicDFU dynamic linking?

mateusforgi, 2023-02-08:
Can we change the linking type to .dynamic?
`.library(name: "NordicDFU", type: .dynamic, targets: ["NordicDFU"])`

I have a XCFramework, that depends on NordicDFU but it cannot link to NordicDFU at runtime because NordicDFU is not linked with dynamic type. 

On apple documentation it says this:
`Leave this parameter so Swift Package Manager can choose between static or dynamic linking (recommended).`

But on the other part it says this:
`A library’s product can be either statically or dynamically linked. It’s recommended that you don’t explicity declare the type of library, so Swift Package Manager can choose between static or dynamic linking based on the preference of the package’s consumer.`

It is kind of confusing as it recommends the dynamic linking but It also says to let the type as nil.
https://developer.apple.com/documentation/packagedescription/product/library(name:type:targets:)

mateusforgi, 2023-02-08:
@philips77 can you take a look on that?

ClaudiaGNM, 2023-02-23:
Same problem here. Please have a look at this!
We are trying to integrate NordicDFU inside a framework distributed via CocoaPod and it will crash on start during runtime.

dinesharjani, 2023-02-23:
So, if you change the library to .dynamic, does this work for you? It fixes the issue?

philips77, 2023-02-23:
Are you referring to this line: 
https://github.com/NordicSemiconductor/IOS-DFU-Library/blob/22952128c6499740f0d04b3dba9b51813cc19a4c/Package.swift#L17

But we don't specify that parameter, so SPM should be able to link it any way you want, right?

mateusforgi, 2023-02-23:
> So, if you change the library to .dynamic, does this work for you? It fixes the issue?

Correct, but we also need to make ZIPFoundation dynamic. I already opened a issue there.

> Are you referring to this line:
> 
> https://github.com/NordicSemiconductor/IOS-DFU-Library/blob/22952128c6499740f0d04b3dba9b51813cc19a4c/Package.swift#L17
> 
> But we don't specify that parameter, so SPM should be able to link it any way you want, right?

Yes, it is this line. We need to change to. 
```swift 
.library(name: "NordicDFU", type: .dynamic, targets: ["NordicDFU"]),
```
But it will not work unless ZIPFoundation is also dynamic. 

Regarding SPM should be able to link it anyway. This is the real issue. It looks like it does not. According to [this comment](https://forums.swift.org/t/issue-with-third-party-dependencies-inside-a-xcframework-through-spm/41977/5) on swift forum there is no way of influence that beyond package manifest.

dinesharjani, 2023-02-23:
@mateusforgi as soon as ZIPFoundation makes their change, either make a PR here or let us know and we'll change it as well.

mateusforgi, 2023-02-23:
@dinesharjani thanks, I will open a pr as soon as they make their change.

philips77, 2023-02-23:
No, no.. that change on our side would force the lib to be dynamically linked. Now it can be linked dynamically or staticly, depending on settings in your project's SPM config. 
We would only change if our lib had to be dynamically linked, but I guess that's not the case

philips77, 2023-02-23:
I guess you're referring to https://stackoverflow.com/a/70812112 which says that xcode always prefers static linking, right? Unless we specify otherwise.