nordicsemi/IOS-DFU-Library #503: Module name with SPM vs Cocoapods

jdanthinne, 2023-02-15:
Is there a reason why this library, when used with Cocoapods, must be imported with `import iOSDFULibrary`, and when used with SPM, must be imported with `import NordicDFU`?

I'm building a SDK that I'd like to offer for Cocoapods and SPM, and having two different names makes it impossible.

philips77, 2023-02-23:
I see the following options:
1. Adding some #ifdefs on your side - is it possible?
2. Renaming our lib on Cocoapods to `NordicDFU` - perhaps other libraries also should be renamed?
3. Deprecating Cocoapods?
4. Saying that we're busy with lots of tasks and will come back to you in the future ;)

I vote for option 2. The current `iOSDFULibrary` can be modified to only include the new one, like a proxy, or can be deprecated. After all *Find & Replace* finally works in Xcode.

jdanthinne, 2023-02-23:
This is what I had to do in the end.

```swift
#if SWIFT_PACKAGE
    import NordicDFU
#else
    import iOSDFULibrary
#endif
```

ehmjaysee, 2024-03-22:
Just FYI. 

I just upgraded to 4.15.0 and it appears the original package name 'iOSDFULibrary' no longer works with CocoaPods. The solution is to use the new name:

import NordicDFU

philips77, 2024-03-22:
Yes, this has changed in 4.14 and the migration guide is here: https://github.com/NordicSemiconductor/IOS-DFU-Library/releases/tag/4.14.0

closed 2024-03-22 by philips77