nordicsemi/IOS-DFU-Library #507: Difference between DfuViewModel and DFUViewController and use of DFUFirmwareProvider

xaviergspintly, 2023-08-28:
I have a question which may have a very simple answer.
I am trying to replicate the example code in a basic project of mine, but, I see that there is repetition of the core logic that I have to use in the files DfuViewModel and DFUViewController.
Which of these files should I actually reference? I feel it is DFUViewController, but please can you explain me the use of both these files in your example project.
I couldn't find any linking between them and in what context are they used separately if so.

Also wanted to know how the DFUFirmwareProvider is used since it is not mentioned in the documentation at:
https://github.com/NordicSemiconductor/IOS-DFU-Library/blob/main/documentation.md
The DFUFirmwareProvider implementation seems important for getting it working but the docs don't explain its use.

xaviergspintly, 2023-08-28:
Ha, OK! So, is DFUFirmwareProvider used to only load a test set of firmware? Or is it really part of the library? Can it be just replaced with my firmware zip?

philips77, 2023-08-28:
Hello,
You need to create a `DFUFrimware` instance and pass it to the `DFUServiceInitiator`.
The `DFUFirmwareProvider` is just a class used in the test app to provide a firmware for each test, that's why it's not documented.

And yes, you need to download the file locally, I believe.

xaviergspintly, 2023-08-29:
@philips77 I got that part of DFUFrimware and DFUServiceInitiator. But why are there 2 similar implementations in DfuViewModel and DFUViewController. What is the difference between these two files which have similar code?

philips77, 2023-08-29:
@sylwester-zielinski could you help? I don't know the details.

sylwester-zielinski, 2023-08-29:
@xaviergspintly I can't find the class DFUViewController. Can you provide a link to it?

xaviergspintly, 2023-08-29:
https://github.com/NordicSemiconductor/IOS-DFU-Library/blob/main/Example/iOSDFULibrary/View%20Controllers/DFUViewController.swift

closed 2023-11-30 by philips77