nordicsemi/IOS-DFU-Library #511: localization

albertopeam, 2023-09-04:
**Your question**
Using the `DFUServiceDelegate` the function `@objc func dfuError(_ error: DFUError, didOccurWithMessage message: String)` returns the message always in EN.
Do you plan to localize the texts for `DFUError` of you think that clients must do it?

**Logs**
-

dinesharjani, 2023-09-04:
Hi @albertopeam - no, at the moment that is not one of our plans. We understand your frustration since it means you might need to localise these errors yourself and keep up to date with library changes. But we've walked back on our attempts to localise everything, since most of our work is consumed by developers, who are used to working on everything in English.

dinesharjani, 2023-09-04:
As for whether clients must localise it, probably depends in the use case. If this is a developer tool you're working on, you should be fine in passing through our own errors and showing their description. If this is a user-facing product or app however, you might need to do some work over there.

closed 2023-09-04 by dinesharjani

albertopeam, 2023-09-04:
Thanks for the clarification :)

philips77, 2023-09-08:
What you may do, is extend the `DFUError` type and `DFURemoteError` with `CustomStringConvertible` and return `NSLocalizedString` and use this one instead of that text.

Have an example here: 
https://github.com/NordicSemiconductor/IOS-nRF-Mesh-Library/blob/3c6cde5f07929872d11542dd27441776bf697580/nRFMeshProvision/Mesh%20Messages/ConfigMessage.swift#L405-L430