nordicsemi/IOS-DFU-Library #512: Privacy Manifest

mateusforgi, 2023-09-15:
**Information**
Apple has announced the[ Privacy Manifest](https://developer.apple.com/documentation/bundleresources/privacy_manifest_files/describing_data_use_in_privacy_manifests), required by apps and SDKs. 

```Third-party SDKs need to provide their own privacy manifest files that record the types of data they collect. Your app’s privacy manifest file doesn’t need to cover data collected by third-party SDKs that your app links to.```

We develop a SDK that has the iOS-DFU-Library as its dependencies, we are already providing the Privacy Manifest for our SDK, however for the future iOS-DFU-Library needs to also have its own file.

**Your question**
When are you guys planning to include it on the iOS-DFU-Library?

dinesharjani, 2023-09-15:
@mateusforgi we will do that, thanks for the reminder. Sometimes we receive a wave of things we need to check, fix or look at immediately and things slip. Sorry about that, we'll look into that on Monday.

mateusforgi, 2023-09-15:
@dinesharjani Thanks. We have some time until it becomes mandatory. 

According to apple: 
```From Fall 2023 you’ll receive an email from Apple if you upload an app to App Store Connect that uses required reason API without describing the reason in its privacy manifest file. From Spring 2024, apps that don’t describe their use of required reason API in their privacy manifest file won’t be accepted by App Store Connect.```

dinesharjani, 2023-09-18:
@mateusforgi I made this PR. Could you have a look? https://github.com/NordicSemiconductor/IOS-DFU-Library/pull/513

philips77, 2023-09-18:
Hi, I went though the list of all privacy usages listed here: https://developer.apple.com/documentation/bundleresources/privacy_manifest_files/describing_data_use_in_privacy_manifests and the restricted API listed here: https://developer.apple.com/documentation/bundleresources/privacy_manifest_files/describing_use_of_required_reason_api and I find nothing that we're using in the DFU library.

I found several uses of the API in *ZipFoundation* library, which the DFU library is using under the hood, e.g.: https://github.com/search?q=repo%3Aweichsel%2FZIPFoundation%20modificationDate&type=code
but those are not exposed and not used by the DFU library.

In my opinion we can just use `3B52.1` for File Timestamp APIs, to be on the safe side.

mateusforgi, 2023-09-19:
@dinesharjani looks good to me.

mateusforgi, 2023-09-26:
@dinesharjani @philips77 ZIPFoundation already made the changes and put in on the develop branch, see thread [here](https://github.com/weichsel/ZIPFoundation/issues/285). However they are using it as a resource file, however on my tests Xcode is not generating the privacy report combining the manifests of the SDKs, it only takes into account the main app manifest.  I am following the threads on Firebase library and Facebook to see how they will handle it.

dinesharjani, 2023-09-27:
@mateusforgi appreciate it. If you could keep us updated, that'd be great.

mateusforgi, 2023-09-27:
@dinesharjani @philips77 I confirmed. We need to add the privacy report as resource file, like ZIPFoundation is doing [here](https://github.com/weichsel/ZIPFoundation/blob/development/Package%40swift-5.9.swift#L7).

philips77, 2023-09-28:
Thanks, we'll add it.

closed 2023-11-30 by philips77