nordicsemi/IOS-DFU-Library #514: DFU progress stuck on iOS 17.0

LucaDonetti, 2023-10-19:
DFU Bootloader version (please complete the following information):

SDK version: NRF5, v16
- Bonding used: yes
- Library version: [4.13.0]
Device information (please complete the following information):

Device: iPhone XS, iPhone 14 Pro, iPad mini 5th Generatoion
- OS: iOS 17.0
Describe the bug:
DFU can be initiated without any problem, how ever the progress stuck at random progress. Only happens on iOS 17.0. Everything is working fine for iOS 16 and below.
Suspect that's a bug from Apple side on iOS 17.0, but might as well post here.
The Bug is very similar to https://github.com/NordicSemiconductor/IOS-DFU-Library/issues/510.
I try to set "packetReceiptNotificationParameter" to 0(like you explain on this report  https://github.com/NordicSemiconductor/IOS-DFU-Library/issues/510), but is not work.

Log:

"[BSDfuOta] [Nordic-Service] State changed: Connecting"
"[BSDfuOta] [Nordic-Service] State changed: Starting"
"[BSDfuOta] [Nordic-Service] State changed: Uploading"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 0 % - Speed: 0.0 - Avg speed: 0.0"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 1 % - Speed: 6958.366797646767 - Avg speed: 6958.366797646767"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 2 % - Speed: 4509.398199166779 - Avg speed: 5473.2088788075325"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 3 % - Speed: 4081.0547312089516 - Avg speed: 4914.6331591716735"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 4 % - Speed: 6529.697413215871 - Avg speed: 5236.988559958262"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 5 % - Speed: 4543.566837350083 - Avg speed: 5081.746769250404"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 6 % - Speed: 4935.882293024704 - Avg speed: 5056.822899650668"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 7 % - Speed: 6722.5135078699495 - Avg speed: 5241.577580605801"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 8 % - Speed: 4308.121523592575 - Avg speed: 5103.192135303826"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 9 % - Speed: 4352.404008856006 - Avg speed: 5007.117919232876"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 10 % - Speed: 5960.438633141871 - Avg speed: 5088.16772177767"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 11 % - Speed: 4150.61427948979 - Avg speed: 4985.644796943376"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 12 % - Speed: 3638.132737776432 - Avg speed: 4836.181889448011"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 13 % - Speed: 3596.889887365924 - Avg speed: 4711.958910960532"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 14 % - Speed: 6161.176733136156 - Avg speed: 4792.716572083866"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 15 % - Speed: 3784.7014997663578 - Avg speed: 4708.986727917955"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 16 % - Speed: 3969.114815418525 - Avg speed: 4655.038745743609"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 17 % - Speed: 7788.733878047134 - Avg speed: 4768.222914186186"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 18 % - Speed: 4505.52774625675 - Avg speed: 4752.804971497896"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 19 % - Speed: 4382.366392777505 - Avg speed: 4731.863693777893"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 20 % - Speed: 7678.367097694481 - Avg speed: 4824.7149159199425"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 21 % - Speed: 4046.672708164637 - Avg speed: 4780.8759847796455"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 22 % - Speed: 4494.97636887608 - Avg speed: 4767.0736768621655"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 23 % - Speed: 3657.189160188722 - Avg speed: 4705.317471268779"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 24 % - Speed: 8387.021503974249 - Avg speed: 4793.250629186246"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 25 % - Speed: 4135.09154503794 - Avg speed: 4762.881458766034"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 26 % - Speed: 3797.4188327619404 - Avg speed: 4717.000045096783"
"[BSDfuOta] [Nordic-Progress] Parts: 1 / 1 - Progress: 27 % - Speed: 8023.633665028591 - Avg speed: 4790.337201423899"
"[BSDfuOta] [Nordic-Service] State changed: Connecting"
"[BSDfuOta] [Nordic-Service] State changed: Starting"
"[BSDfuOta] [Nordic-Service] Error 305: Writing to characteristic failed"

zwift-aebling, 2023-10-20:
Can confirm we are seeing similar behaviour here, which was not reproducible until we updated the device to iOS 17.

LucaDonetti, 2023-10-23:
Any solutions? Because some clients that update to iOS 17 can't update to the new firmware

JtheSaw, 2023-10-23:
Some of my clients are experiencing the same problem but I can't reproduce it with my test devices. Do you have a way to reliably force the problem to occur?

philips77, 2023-10-24:
Hello, I'll try to look into this issue this week. I need to finish my current task before. Sorry that I keep you waiting.

jpszczerba, 2023-11-17:
Hey, do you have any update regarding this?

philips77, 2023-11-17:
I'm resuming work on this project just now.

jpszczerba, 2023-11-17:
Okay, thank you!

Yramani1998, 2023-11-27:
Still facing the same issue on iphone 12 (ios 17.0.0) . Do we have any updates ?

philips77, 2025-04-02:
Please specify the nRF5 SDK version, are you using bonding or not and share logs. I just tried DFU on iOS 18.3.2 on an iPhone 13 and it worked.

brix-nobo, 2025-04-02:
We're using an nRF52833 development kit.

We're not using the iOS DFU library directly, but rather in a MAUI app via the `Laerdal.Dfu` and `Laerdal.Dfu.Bindings.iOS` packages.

The problem is exactly as described above: The download starts, but at a certain random point in time, we no longer receive any status or progress updates.
Even if it appears that the update is frozen, it will still run and complete in the background.

We would appreciate any help!

philips77, 2025-04-02:
I would suggest enabling Packet Receipt Notifications (PRN) and setting the value to 6 or lower it it continues to fail. 
I hope the bindings allow you to do this.

brix-nobo, 2025-04-02:
Thank you so far, I will check this