nordicsemi/IOS-DFU-Library #526: nRF52840  DFU cannot find service

duansenlin123, 2024-01-17:
- Bonding used: YES
 - Library version: 4.14.0

 - Device: iPhone 12  
 - OS: iOS 16.2

**question**

When upgrading to nRF52840, an error message appears indicating that the service cannot be found

**Logs**

0gWith ld: [Callback] Central Manager did update state to: Powered ON
1gWith ld: Connecting to Matter25Q...
0gWith ld: centralManager.connect(peripheral, options: nil)
0gWith ld: [Callback] Central Manager did connect peripheral
5gWith ld: Connected to Matter25Q
1gWith ld: Discovering services...
0gWith ld: peripheral.discoverServices(nil)
5gWith ld: Services discovered
24gWith ld: DFU Service not found
0gWith ld: The following services were discovered:
0gWith ld:  - 00010203-0405-0607-0809-0A0B0C0D1910
0gWith ld:  - 8D53DC1D-1DB7-4CD3-868B-8A527460AA84
0gWith ld: Did you connect to the correct target? It might be that the previous services were cached: toggle Bluetooth from iOS settings to clear cache. Also, ensure the device contains the Service Changed characteristic
1gWith ld: Disconnecting...
0gWith ld: centralManager.cancelPeripheralConnection(peripheral)
0gWith ld: [Callback] Central Manager did disconnect peripheral
5gWith ld: Disconnected

dinesharjani, 2024-01-17:
Dumb question but, did you follow what the logs themselves suggest? That you clear the iPhone's Bluetooth cache by going to Settings -> Bluetooth -> Off and then On? Using Control Panel does not work, because it doesn't turn Bluetooth off but just stop advertising instead.

duansenlin123, 2024-01-19:
All done, it doesn't work

duansenlin123, 2024-01-19:
> Dumb question but, did you follow what the logs themselves suggest? That you clear the iPhone's Bluetooth cache by going to Settings -> Bluetooth -> Off and then On? Using Control Panel does not work, because it doesn't turn Bluetooth off but just stop advertising instead.

All done, it doesn't work

philips77, 2024-02-19:
Well... looks like your device does not support Legacy or Secure DFU or the UUID aren't standard. Using the `UuidHelper` it is possible to change the service UUIDs if necessary.

closed 2024-02-19 by philips77