nordicsemi/IOS-DFU-Library #537: Intermittent Error 8: Operation Not Permitted when doing update for multiple devices simultaneously

knakul853, 2024-07-26:
With iPhone 15, When updating multiple BLE devices simultaneously, we have observed an intermittent `Error 8: Operation not permitted` issue.
The firmware update zip package has a size of around 360kb.  The error usually happens near the start but sometimes happens after several packets are transferred. The DFU secure_bootloader comes from SDK 15.2.0_9412b96 and we do NOT use bonding.  We have captured a log with verbosity as below provided.

We found a similar issue  https://github.com/NordicSemiconductor/IOS-DFU-Library/issues/319 and tried to change packetReceiptNotifications from 1-22 and all values resulted in failures.

our code for initialization
`    

     guard let pInfo = BluetoothManager.peripheralInfo[peripheral.identifier] else {
            logger.log(level: .error, message: "Peripheral info not found for peripheral: \(peripheral.identifier)")
            return
        }
        guard let firmware = pInfo.selectedFirmware else {
            logger.log(level: .error, message: "No selected firmware for peripheral: \(peripheral.identifier)")
            pInfo.dfuUpdateDidComplete(success: false)
            return
        }
        
        logger.log(level: .info, message: "Starting DFU update with peripheral: \(peripheral.identifier)")

        let initiator = DFUServiceInitiator().with(firmware: firmware)
        initiator.logger = pInfo
        initiator.delegate = pInfo
        initiator.progressDelegate = pInfo
        initiator.alternativeAdvertisingNameEnabled = false
        initiator.disableResume = true

        let _ = initiator.start(target: peripheral)

`

**DFU Bootloader version **
 - SDK version: 15.2.0_9412b96 
 - Bonding used: No
 - Library version: 4.15.1

**Device information **
 - Device: iPhone 15
 - OS: iOS 18

**Question**
With this library is it allowed to start multiple devices simultaneously for firmware update?
When multiple devices are set to update simultaneously  does it impact the message flow, Even though `packetReceiptNotifications` parameter we tried with from 1-20  all resulted into `Error 8: Operation not permitted` error.

**Log**

[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] [DFU State Change] - Connecting
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- [Callback] Central Manager did update state to: Powered ON
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Connecting to gse-1000520842...
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- centralManager.connect(peripheral, options: nil)
the percentage updated:  0.0 number of devices finished update:  0 firmware update stattus complete: false
[debug] [AC119ECC-29DC-B020-A2F9-9FC19885A56A] DFU Update Log --- [Callback] Central Manager did connect peripheral
[info] [AC119ECC-29DC-B020-A2F9-9FC19885A56A] DFU Update Log --- Connected to gse-3365108
[debug] [AC119ECC-29DC-B020-A2F9-9FC19885A56A] DFU Update Log --- Discovering services...
[debug] [AC119ECC-29DC-B020-A2F9-9FC19885A56A] DFU Update Log --- peripheral.discoverServices(nil)
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- [Callback] Central Manager did connect peripheral
[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Connected to gse-1000520842
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Discovering services...
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- peripheral.discoverServices(nil)
the percentage updated:  0.0 number of devices finished update:  0 firmware update stattus complete: false
[info] [AC119ECC-29DC-B020-A2F9-9FC19885A56A] DFU Update Log --- Services discovered
[debug] [AC119ECC-29DC-B020-A2F9-9FC19885A56A] DFU Update Log --- Starting Secure DFU...
[info] [AC119ECC-29DC-B020-A2F9-9FC19885A56A] DFU Update Log --- Connected to gse-3365108
[info] [AC119ECC-29DC-B020-A2F9-9FC19885A56A] DFU Update Log --- Services discovered
[debug] [AC119ECC-29DC-B020-A2F9-9FC19885A56A] DFU Update Log --- Secure DFU Service found
[debug] [AC119ECC-29DC-B020-A2F9-9FC19885A56A] DFU Update Log --- Discovering characteristics in DFU Service...
[debug] [AC119ECC-29DC-B020-A2F9-9FC19885A56A] DFU Update Log --- peripheral.discoverCharacteristics(nil, for: FE59)
[info] [AC119ECC-29DC-B020-A2F9-9FC19885A56A] DFU Update Log --- DFU characteristics discovered
[info] [AC119ECC-29DC-B020-A2F9-9FC19885A56A] [DFU State Change] - Starting
[debug] [AC119ECC-29DC-B020-A2F9-9FC19885A56A] DFU Update Log --- Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
[... 202 lines cut ...]
the percentage updated:  0.0 number of devices finished update:  0 firmware update stattus complete: false
the percentage updated:  0.0 number of devices finished update:  0 firmware update stattus complete: false
the percentage updated:  0.0 number of devices finished update:  0 firmware update stattus complete: false
the percentage updated:  0.0 number of devices finished update:  0 firmware update stattus complete: false
[debug] [AC119ECC-29DC-B020-A2F9-9FC19885A56A] DFU Update Log --- [Callback] Central Manager did disconnect peripheral
[info] [AC119ECC-29DC-B020-A2F9-9FC19885A56A] DFU Update Log --- Disconnected by the remote device
[info] [AC119ECC-29DC-B020-A2F9-9FC19885A56A] DFU Error --- Operation not permitted%
[info] [AC119ECC-29DC-B020-A2F9-9FC19885A56A] Bluetooth Peripheral - dfuUpdateDidComplete success: false
the show load value is: false
[warn] ScannedPeripheral: DFU Update failed for peripheral - AC119ECC-29DC-B020-A2F9-9FC19885A56A
[error] Error 18: Operation not permitted
XPC connection invalid
DeviceUpdateTrackingManager: AC119ECC-29DC-B020-A2F9-9FC19885A56A
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- [Callback] Central Manager did disconnect peripheral
XPC connection invalid
[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Disconnected by the remote device
[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Error --- Operation not permitted%
[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] Bluetooth Peripheral - dfuUpdateDidComplete success: false
the show load value is: false
[warn] ScannedPeripheral: DFU Update failed for peripheral - E3F263BF-5520-88F1-5901-1850D1061C0F
[error] Error 18: Operation not permitted
the percentage updated:  1.0 number of devices finished update:  2 firmware update stattus complete: true
[info] all device update completed.
DeviceUpdateTrackingManager: E3F263BF-5520-88F1-5901-1850D1061C0F
[info] update completed successfully, stoping timer.

philips77, 2024-07-27:
Hello,
It looks like you're trying to update the same device (AC119ECC-29DC-B020-A2F9-9FC19885A56A) twice simultaneously.
I see there are duplicated logs for that device, as if you'd stared it twice, and the device sends "operation not permitted" at some point, when the second instance changes the state for the second time.

For example, you can find the following line twice:
```
[info] [AC119ECC-29DC-B020-A2F9-9FC19885A56A] DFU Update Log --- Connected to gse-3365108
```

philips77, 2024-07-27:
The 'centralManager.connect(peripheral, options: nil)` is called once, so seems like it is the system that calls `perpheral(peripheral: didConnect...` twice for the device. This is intersting...

philips77, 2024-07-27:
From the logs, this seconds log appears to be even before you start connecting to the other device.

philips77, 2024-07-27:
Seems like some logs are missing from the beginning when you connect to that device:
```
[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] [DFU State Change] - Connecting
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- [Callback] Central Manager did update state to: Powered ON
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Connecting to gse-1000520842...
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- centralManager.connect(peripheral, options: nil)
the percentage updated:  0.0 number of devices finished update:  0 firmware update stattus complete: false
[debug] [AC119ECC-29DC-B020-A2F9-9FC19885A56A] DFU Update Log --- [Callback] Central Manager did connect peripheral
[info] [AC119ECC-29DC-B020-A2F9-9FC19885A56A] DFU Update Log --- Connected to gse-3365108
```
There's no `centralManager.connect(...)` to AC119ECC-29DC-B020-A2F9-9FC19885A56A.

knakul853, 2024-07-28:
@philips77  update the logs with initial logs, We checked with a single device also, some logs are repeated and we can see `DFU Update Log --- Connected to gse-xxxxxxx` twice  eg:

this is a single device update log:

[info] device gse-10005208425 is selected for firmware update
DeviceUpdateTrackingManager: E3F263BF-5520-88F1-5901-1850D1061C0F
[info] started updaing device: gse-10005208425
[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] Starting DFU update
[info] Starting DFU update with peripheral: E3F263BF-5520-88F1-5901-1850D1061C0F
[info] ScannedPeripheral: DFU Update started
DeviceUpdateTrackingManager: E3F263BF-5520-88F1-5901-1850D1061C0F
[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] [DFU State Change] - Connecting
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- [Callback] Central Manager did update state to: Powered ON
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Connecting to gse-1000520842...
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- centralManager.connect(peripheral, options: nil)

[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- [Callback] Central Manager did connect peripheral
[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Connected to gse-1000520842
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Discovering services...
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- peripheral.discoverServices(nil)
[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Services discovered
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Starting Secure DFU...
[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Connected to gse-1000520842
[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Services discovered
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Secure DFU Service found
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Discovering characteristics in DFU Service...
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- peripheral.discoverCharacteristics(nil, for: FE59)
[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- DFU characteristics discovered
[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] [DFU State Change] - Starting
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50
[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Buttonless DFU indications enabled
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Application with buttonless update found
[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] [DFU State Change] - Enabling DFU Mode
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- peripheral.writeValue(0x01, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50
[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200101
[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Response (Op Code = Enter Bootloader, Status = Success) received
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- [Callback] Central Manager did disconnect peripheral
[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Disconnected by the remote device
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Scanning for the DFU Bootloader...
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- centralManager.scanForPeripherals(withServices, [FE59])

[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- DFU Bootloader found with name IoTRecovery3
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Connecting to IoTRecovery3...
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- centralManager.connect(peripheral, options: nil)

[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- [Callback] Central Manager did connect peripheral
[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Connected to IoTRecovery3
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Discovering services...
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- peripheral.discoverServices([FE59])
[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Services discovered
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Secure DFU Service found
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- Discovering characteristics in DFU Service...
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- peripheral.discoverCharacteristics(nil, for: FE59)
[info] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- DFU characteristics discovered
[debug] [E3F263BF-5520-88F1-5901-1850D1061C0F] DFU Update Log --- MTU set to 247
[... cut ...]
DeviceUpdateTrackingManager: E3F263BF-5520-88F1-5901-1850D1061C0F
XPC connection invalid
the percentage updated:  1.0 number of devices finished update:  1 firmware update stattus complete: true
[info] all device update completed.
[info] update completed successfully, stoping timer.
[info] Bluetooth scan: Start
[info] scanningStatusChanged  to true
[info] start scanning animation
removing devices
[info] [AC119ECC-29DC-B020-A2F9-9FC19885A56A] New Peripheral gse-33651084 found. Adding to scanner list view. Name

jefferypalmer, 2024-08-01:
@philips77 Our team can / will share more information, but we have clearly narrowed it down to an issue with this library. To eliminate any possibility of app related issues, we coded up a simple structure to initiate an update on multiple devices simultaneously with no delegates, single function, very basic library initiation. Single device update works perfect every time. Multi device update by calling DFUManager.update() twice and allowing the library to run through the update process fails almost every time. The library seems to start connecting / confusing the other device being updated, initiating connects multiple times, etc. None of these issues exist for a single device update.

Logs/details are important, will fill in any details you need, but have a quick question to narrow down some quick validation testing:

1) Does the "nRF DFU" app allow multi device update? It seems the UI is really designed to only allow selecting a single device
2) Is there existing tests in this library for multi device updates? If so, can you possibly point us to where those tests are or any pointers to some of the multi-device update handling code?

philips77, 2024-08-01:
Thank you for the logs. I'll review then as soon as possible. I never tried multi device update, but every update should be independent and not interfere with other instances. Of course there's a point where the lib meets the os calls and I suspect a problem there. As you wrote , the lib may mistakenly respond to start calls from all running instances, not just the one it should. I spent some times in your previous logs, trying to find that place, but the interesting part was missing. With more logs I can go further. 

1. Yes, only single device mode.
2. We don't have such.

knakul853, 2024-08-01:
@philips77  Here are Some logs that might be useful for your debugging

1.  2 devices selected for firmware, the update process starts both of the updates together, Where both of the device failed with error `Error 8: Operation not permitted` We get 80-90% of the time this failure
[Dual Update Failure.txt](https://github.com/user-attachments/files/16458411/Dual.Update.Failure.txt)

2. Sometime it gets fails whyilematching CRC  and doesn't go beyond but it happens very few time. We tried different value of RPN but that doesn't help
[DualUpdateFailureCRCDoesn'tMatch.txt](https://github.com/user-attachments/files/16458465/DualUpdateFailureCRCDoesn.tMatch.txt)

3. There are some success cases where multiple devices gets updated successfully
[Dual-Update Success.txt](https://github.com/user-attachments/files/16458490/Dual-Update.Success.txt)

jefferypalmer, 2024-08-01:
@philips77 to add some context to these logs above, you will see in the logs with 2 updates initiated (labelled Dual) that there are multiple connection attempts and subsequent failures. For single device update (same code, just a single device selected, nakul is posting now) there are not multiple connection attempts, and always succeeds in programming.

Wichnator, 2024-08-13:
Hello, I have just one suggestion. Did you try to synchronize "connecting/initial" stage of the DFU process?
I had an issue with multi-device update (limited to 5 in parallel) and my issue was that some device were "sometimes" disconnected or I got an error: "error=15, message=Invalid object".
A similar issue was also on Android.

I added synchronization like: Do not start another update until the current one is not at least in DFUState.uploading stage.