nordicsemi/IOS-DFU-Library #542: Not working with AccessorySetupKit (iOS18+)

jdanthinne, 2024-10-15:
**DFU Bootloader version (please complete the following information):**
 - SDK version: ?
 - Bonding used: no
 - Library version: 4.15.3

**Device information (please complete the following information):**
 - Device: iPhone 16 Pro
 - OS: iOS 18.0

**Describe the bug**
Update don't start with a device paired with AccessorySetupKit (iOS18).

**Logs**
2024-10-15T09:02:54.997Z [CLOTANordic] [Firmware Update] DFU state: Connecting
🚨 2024-10-15T09:02:54.998Z [CLOTANordic] [Firmware Update] DFU error: Bluetooth adapter powered off. DFUError
🚨 2024-10-15T09:02:55.008Z [CLOTA] [DBP_dc35] [OTA Update] Update did fail. updateFailure(Error Domain=DFUError Code=203 "Bluetooth adapter powered off" UserInfo={NSLocalizedDescription=Bluetooth adapter powered off})

philips77, 2024-10-15:
> DFU error: Bluetooth adapter powered off. DFUError

Hmm... Do you have Bluetooth enabled?

jdanthinne, 2024-10-15:
Yes, but CoreBluetooth behaves differently when using AccessorySetupKit. I've cloned this repo and removed the error when CBCentralManager is off, because with ASK, state is always off at first… but now, it is stuck when trying to connect the DFU device, because it has not been paired with ASK. I don't see how it could be possible now 😬

```swift
DFU log: [Callback] Central Manager did update state to: Powered OFF
DFU log: [Callback] Central Manager did update state to: Powered ON
DFU log: Connecting to DBP_dc35...
DFU log: centralManager.connect(peripheral, options: nil)
DFU log: [Callback] Central Manager did connect peripheral
DFU log: Connected to DBP_dc35
DFU log: Discovering services...
DFU log: peripheral.discoverServices(nil)
DFU log: Services discovered
DFU log: Starting Secure DFU...
DFU log: Connected to DBP_dc35
DFU log: Services discovered
DFU log: Secure DFU Service found
DFU log: Discovering characteristics in DFU Service...
DFU log: peripheral.discoverCharacteristics(nil, for: FE59)
DFU log: DFU characteristics discovered
DFU state: Starting
DFU log: Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...
DFU log: peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)
DFU log: Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50
DFU log: Buttonless DFU indications enabled
DFU log: Application with buttonless update found
DFU state: Enabling DFU Mode
DFU log: Trying setting bootloader name to Dfu38507
DFU log: Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
DFU log: peripheral.writeValue(0x02084466753338353037, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
DFU log: Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50
DFU log: Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200201
DFU log: Response (Op Code = Set Name, Status = Success) received
DFU log: Bootloader name changed successfully
DFU log: Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...
DFU log: peripheral.writeValue(0x01, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)
DFU log: Data written to 8EC90003-F315-4F60-9FB8-838830DAEA50
DFU log: Indication received from 8EC90003-F315-4F60-9FB8-838830DAEA50, value (0x):200101
DFU log: Response (Op Code = Enter Bootloader, Status = Success) received
DFU log: [Callback] Central Manager did disconnect peripheral
DFU log: Disconnected by the remote device
DFU log: Scanning for the DFU Bootloader...
DFU log: centralManager.scanForPeripherals(withServices, [FE59])
DFU log: Scanning timed out returning no matching peripherals!
DFU log: centralManager.stopScan()
DFU error: No DFU device found.. DFUError
```

philips77, 2024-10-15:
I think you need to send a command to switch to DFU bootloader mode using ASK, and then scan and run the DFU lib for the scanned device.

Have a look at nRF Toolbox app: https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/tree/master/nRF%20Toolbox/Profiles/DeviceFirmwareUpdate/ViewConrollers/HomeKit
You may download the app from AppStore and try if it works for you.

jdanthinne, 2024-10-15:
I don't see how to _send a command to switch to DFU bootloader mode using ASK_. ASK is not a replacement for CoreBluetooth, it only allows pairing devices, not sending BLE commands.

philips77, 2024-10-15:
Have a look at: https://github.com/NordicSemiconductor/IOS-nRF-Toolbox/blob/4d071a82fd39241cc8a376058c8033fd1ff54597/nRF%20Toolbox/Profiles/DeviceFirmwareUpdate/ViewConrollers/HomeKit/HMAccessoryListTableViewController.swift#L227

`HMAccessory` has `services`, etc.

jdanthinne, 2024-10-15:
You're talking about HomeKit. I'm talking about [AccessorySetupKit](https://developer.apple.com/documentation/accessorysetupkit/), which is a different beast…

philips77, 2024-10-15:
Oh, that's iOS 18+. Too fresh for me.

Anyway. It depends on the type of the DFU bootloader and what do you want to update. For the ASK to work, the device must have the same MAC address in app mode and in bootloader mode. This may be a case when the device is bonded and you're using `8EC90004-F315-4F60-9FB8-838830DAEA50` buttonless service, which restarts the device and sends Service Changed indication to indicate new set of services. However, it works only when the single firmware updated is the application (or perhaps SD, but with minor version). During major SD update the app gets removed, so in order to flash a new app the bootloader advertises with address incremented by 1. The same happens when the device isn't bonded and `8EC90003-F315-4F60-9FB8-838830DAEA50` service is used (the one you're using).

It means, that you are able to connect to the device in app mode using ASK, but not to the bootloader, which, from iPhone's point of view, becomes completely new device (as it has a new MAC address). The library scans for it using CoreBluetooth.

If my limited understanding of Accessory Setup Kit is correct, you won't be able to update a device with nRF5 SDK which has Legacy or Secure DFU service.

If you'd have devices built on nRF Connect SDK or Zephyr, which are using MCU Manager and are not switching to bootloader, the update would have worked.

To sum up, the only case when ASK has option to work with this library (after your modifications) is when the device is bonded and is using Buttonless Service With Bonds, and only when updating the app or bootloader.

jdanthinne, 2024-10-16:
Okay, thanks for the explanation. Unfortunately, that's the conclusion I had reached. I'll try to have a technical contact at Apple to see what's possible (whitelist?), but I don't have many hopes… Many devices we're building are using Nordic DFU and this would be a major drawback.

philips77, 2024-10-16:
Why can't you continue to use CoreBluetooth? Asking user to use Bluetooth only before DFU?

You could also try modifying the DFU bootloader to not use address +1, but instead send Service Change indication (even without bond) and check how it works. But anyway, you'd have to update existing devices using Core Bluetooth and later modify the DFU library for the updated devices.

jdanthinne, 2024-10-16:
The way ASK works is kind of excluding the classic use of CoreBluetooth, it's one or the other, so I think we need to go back to classic CoreBluetooth until we change the DFU bootloader the way you've explained, because many existing devices are already in stores… thanks for all the advices anyway!

closed 2024-10-16 by philips77